ARM-software/CMSIS_5 #126: DSP：is ARM_MATH_CM0_FAMILY_FAMILY a mistake in writing?

xizhizhang, 2016-12-14:
hi,
There are some ARM_MATH_CM0_FAMILY_FAMILY in DSP/Source/TransformFunctions/. I wonder, if they should be ARM_MATH_CM0_FAMILY?

> CMSIS_5/CMSIS$ grep  CM0_FAMILY_FAMILY DSP/ -nr
> DSP/Source/TransformFunctions/arm_cfft_radix4_f32.c:79:#ifndef ARM_MATH_CM0_FAMILY_FAMILY
> DSP/Source/TransformFunctions/arm_cfft_radix4_f32.c:557:#endif /* #ifndef ARM_MATH_CM0_FAMILY_FAMILY */
> DSP/Source/TransformFunctions/arm_cfft_radix4_f32.c:583:#ifndef ARM_MATH_CM0_FAMILY_FAMILY
> DSP/Source/TransformFunctions/arm_cfft_radix4_f32.c:1163:#endif /* #ifndef ARM_MATH_CM0_FAMILY_FAMILY */
> DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_q31.c:93:#ifndef ARM_MATH_CM0_FAMILY_FAMILY
> DSP/Source/FilteringFunctions/arm_biquad_cascade_df1_q31.c:397:#endif /*  #ifndef ARM_MATH_CM0_FAMILY_FAMILY */

Thanks

GuentherMartin, 2016-12-14:
Hello xizhizhang,

Thank you for pointing out.
The macros should be ARM_MATH_CM0_FAMILY

GuentherMartin, 2016-12-14:
Updated related source files and rebuild libraries.
Please see https://github.com/ARM-software/CMSIS_5/commit/4442750cff2d3b8b3e352f05ee6c94e8a5948cb8

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil