ARM-software/CMSIS_5 #127: your gcc ASM Scheduling barrier is no scheduling barrier

nolange, 2016-12-19:
You have code in CMSIS/Core/Include/cmsis_gcc.h, function `__set_FPSCR`, and other places where you want to deny reordering instructions around the function. But the empty asm statements dont do that!

```
// arm-none-eabi-g++ -march=armv7-a -mfloat-abi=hard -O2 -c test.cpp

void foo(int *ptr, unsigned long fpscr) {
	*ptr = 7;
  asm volatile ("");                                           /* Empty asm statement works as a scheduling barrier */
  asm volatile ("VMSR fpscr, %0" : : "r" (fpscr) : "vfpcc");
  asm volatile ("");
	*ptr = 8;
	
}
```

will result in code:
```
00000000 <_Z3fooPim>:
   0:	eee11a10 	vmsr	fpscr, r1
   4:	e3a03008 	mov	r3, #8
   8:	e5803000 	str	r3, [r0]
   c:	e12fff1e 	bx	lr
```

Compare that to a proper barrier, code:
```
// arm-none-eabi-g++ -march=armv7-a -mfloat-abi=hard -O2 -c test.cpp

void foo(int *ptr, unsigned long fpscr) {
	*ptr = 7;
  asm volatile ("VMSR fpscr, %0" : : "r" (fpscr) : "vfpcc", "memory");
	*ptr = 8;
	
}
```
result:
```
00000000 <_Z3fooPim>:
   0:	e3a03007 	mov	r3, #7
   4:	e5803000 	str	r3, [r0]
   8:	eee11a10 	vmsr	fpscr, r1
   c:	e3a03008 	mov	r3, #8
  10:	e5803000 	str	r3, [r0]
  14:	e12fff1e 	bx	lr
```

GuentherMartin, 2016-12-21:
Hello nolange,

file cmsis_gcc.h has been updated.
Please see https://github.com/ARM-software/CMSIS_5/commit/9bb952959db584b6f227fa6f0f5ccfe0207fac7f

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil