ARM-software/CMSIS_5 #129: _mutex_*  functions are excluded in RTX2 (armcc)

0xc0170, 2017-01-10:
One of the mbed tests started failing once we updated to RTX2 (mbed malloc tests - https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbedmicro-rtos-mbed/malloc/main.cpp - using threads, each one malloc some memory). RTX does not fail, so this lead me to believe there is a change in RTX2 that is causing this failure. It's only for ARMCC.

It looks like that ``_mutex_`` wrappers are not in the final image. Looking at the changes between RTX and RTX2, these wrapper functions used to be with `` __attribute__((used))`` to be retained. 
- RTX sources of those wrappers : https://github.com/ARM-software/CMSIS_5/blob/4da2d09bb336e1c08ad54fa9988658eccaa8d169/CMSIS/RTOS/RTX/INC/RTX_CM_lib.h#L338. 
- The RTX2 sources of those wrappers: https://github.com/ARM-software/CMSIS_5/blob/4da2d09bb336e1c08ad54fa9988658eccaa8d169/CMSIS/RTOS2/RTX/Include/rtx_config.h#L531

As I understand these symbols are weakly-linked (http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0475m/chr1358938915186.html). However our tests or even a project in the uvision IDE fail (they build object files, do not create an archive), thus should be linked. Debugging shows they are not. if I wrap malloc/free calls plus protect calling their "super" parts with a mutex, the test passes.

Once I add used attribute, the test passes.

Any suggestions?

0xc0170, 2017-01-10:
cc @bulislaw @c1728p9

0xc0170, 2017-01-10:
To reproduce the problem:

Clone mbed-os repository, and run the following command (installed mbed-cli and its test dependencies -  mbed greentea and mbedls and mbedhrun)
```mbed test -n tests-mbedmicro-rtos-mbed-malloc -m K64F -t ARM``.

From master, this test should pass. If you switch to feature_cmsis5 branch on mbed-os repository, the test fails (ERROR reported):

```
+----------+---------------+----------------------------------+----------------------------------+--------+--------+--------+--------------------+
| target   | platform_name | test suite                       | test case                        | passed | failed | result | elapsed_time (sec) |
+----------+---------------+----------------------------------+----------------------------------+--------+--------+--------+--------------------+
| K64F-ARM | K64F          | tests-mbedmicro-rtos-mbed-malloc | tests-mbedmicro-rtos-mbed-malloc | 0      | 1      | FAIL   | 25.42
     |
+----------+---------------+----------------------------------+----------------------------------+--------+--------+--------+--------------------+
```

RobertRostohar, 2017-01-10:
"used" attribute has been added to _mutex_* functions which should resolve the problem.

0xc0170, 2017-01-10:
Thanks @RobertRostohar for fixing this ! This resolves the issue, confirmed running the malloc test we have.

I'll close this issue.

closed 2017-01-10 by 0xc0170