ARM-software/CMSIS_5 #130: PACK.xsd: Cortex-M0+ is invalid name for Pname attribute

oakm, 2017-01-12:
The Pname attribute is mandatory for devices that embed multiple processors. When I use Cortrex-M0+ for processor identifier, the XML validation against PACK.xsd schema fails with the following errors:

ERROR: Element 'processor', attribute 'Pname': [facet 'pattern'] The value 'Cortex-M0+' is not accepted by the pattern '[\-_A-Za-z0-9]+'.
ERROR: Element 'processor', attribute 'Pname': 'Cortex-M0+' is not a valid value of the atomic type 'RestrictedString'.

jkrech, 2017-01-16:
This is an intentional restriction. You may have noticed that in CMSIS CORE we refer the Cortex-M0+ e.g. in defines: __CM0PLUS_REV as Cortex-M0plus instead of using the + sign itself.
Please note that Pname is a reference to a processor instance rather than the processor's name which is captured in Dcore.

So in case you have a dual Cortex-M0+ device you are more likely to choose Pname being e.g. 
core1 and core2
or 
master and slave.
Note that when selecting a device for a multi-core target, you always need to select the processor instance as well.

e.g. in the LPC54000_DFP you will have a device:
LPC54114J256BD64:M0plus
and an 
LPC54114J256DB64:M4
In this case the Pnames chosen are "M0plus" and "M4".

oakm, 2017-01-18:
Agree. Thanks for the prompt response. The issue is resolved.

closed 2017-01-18 by oakm