ARM-software/CMSIS_5 #131: PackChk.exe generates unrelated WARNING when use Pname attribute

oakm, 2017-01-12:
PackChk generates irrelevant warning messages when Pname attribute is used for processor element in the PDSC.

Here is a description example:
 <device Dname="CY8C617FMI-BLD73">
  <processor Pname="Cortex-M4"  Dcore="Cortex-M4"  DcoreVersion="r0p1" Dfpu="1" Dmpu="1" Dendian="Little-endian"/>
  <processor Pname="Cortex-M0+" Dcore="Cortex-M0+" DcoreVersion="r0p1" Dfpu="0" Dmpu="1" Dendian="Little-endian"/>
  <compile   header="devices/psoc6/psoc6ble/include/cy8c617fmi_bld73.h" define="CY8C617FMI_BLD73"/>
  <memory    id="IROM1"         start="0x10000000"  size="0x100000"/>
  <memory    id="IRAM1"         start="0x08000000"  size="0x48000"/>
</device>

**WARNING**
*** WARNING M380: C:\PDL\__3.0.0\Cypress.PDL.pdsc (Line 82) 
  No description found for [Cypress:19] 'CY8C617FMI-BL603'

If I remove Pname element, the warning goes away. But I need it because the device embed 2 processor cores.

oakm, 2017-01-12:
[PDSC.txt](https://github.com/ARM-software/CMSIS_5/files/701269/PDSC.txt)
Attached an example of PDSC from the previous post.

jkrech, 2017-01-16:
Would you mind sending us a pdsc file to reproduce this warning message ?
Have you checked your <description> tags ?
From the warning above it sounds like that there is a device CY8C617FMI-BL603 not having multiple processors, however the system does not find a description for it, as you may have added Pname to your descriptions as well. However this is all speculation until we have a reproducer.

oakm, 2017-01-18:
[Cypress.PDL.txt](https://github.com/ARM-software/CMSIS_5/files/714053/Cypress.PDL.txt)
Attached. The file generates the mentioned warning. If remove Pname attribute, the warning is not generated.
Also, the PackChk tool can be enhanced to generate an error if the description for multi core device does not include Pname attribute.

jkrech, 2017-01-18:
Unfortunately this turns out to be a defect in the PackChk.exe.
a) the description for a device is the concatenation of the <description> tags from family, subfamily, device and in case it exists the variant level.
b) the Pname attribute can be used to make a tag specific for the respective processor instance
(CortexM0p vs. CortexM4)
In your PDSC file you have a description without Pname, which indicates that it will be used for both of your processor instances:
CY8C617FMI-BLD73:CortexM0p
and
CY8C617FMI-BLD73:CortexM4
You can ignore this warning for now.
You may want to add an instance specific description though detailing the benefits that are exclusive to the CortexM0p and CortexM4 respectively. Which would make this warning go away.
The warning may read that there is no exclusive description for:
CY8C617FMI-BLD73:CortexM4
and
CY8C617FMI-BLD73:CortexM0p

jkrech, 2017-01-20:
PackChk V1.3.53 has been committed to the develop branch. This version should fix your problem. Please validate and let us know.

oakm, 2017-01-24:
Confirmed. The defect is fixed. Thanks.

closed 2017-01-24 by oakm