ARM-software/CMSIS_5 #133: Return values of flag functions changed to uint32_t

FerdinandEn, 2017-01-16:
The return values of the flag functions have changed from  `int32_t ` to  `uint32_t `. Before, it was possible to check for error by 

`if(returnval < 0)`

Now I can detect an error if MSB is set. For `int32_t` this is a negative value.
The reason of the error is coded as `osStatus_t`. All error codes of `osStatus_t` are negative values. If I want to decode the error I have to cast the return value from `uint32_t` to `int32_t`.
How should an error detected and decoded:

`If(returnvalue & 0x80000000)
{
  // error and now compare with negative values
}`

I would prefer, if the return values of the flag function are  `int32_t `.

RobertRostohar, 2017-01-18:
Definitions for errors returned by flag functions have been added (osFlagsErrorXxxx). This should avoid the casting.

Return value type has been changed for better MISRA compliance.

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil