ARM-software/CMSIS_5 #134: netInitialize (from Keil Network middleware) fails with NET_ERROR_CONFIG, same code unaffected in API1

kenshin1988, 2017-01-16:


RobertRostohar, 2017-01-17:
This is not really about APIs but rather about RTX configuration required for Network middleware. Most likely not enough memory or stack configured in RTX5. You should enable debug in Network to get more information why Network initialization fails.

RobertRostohar, 2017-01-17:
Cross post: http://www.keil.com/forum/60800/

RobertRostohar, 2017-01-17:
You should check the Keil forum (see previous comment) or contact Keil technical support:http://www.keil.com/support

furbanc, 2017-01-17:
RTOS resource requirements are documented here:
http://www.keil.com/pack/doc/mw/Network/html/nw_resource_requirements.html#nw_rtos_req

mrfuchs, 2017-01-20:
Maybe related: With my project, netInitialize() crashes because of os_svcTimerNew() being called with attr->cb_mem pointing to invalid memory space. In consequence, when the timer control block is tried to be initialized a bus fault occurs. It works like a charm when using RTOS v1.

mrfuchs, 2017-01-23:
For the record, it seems to work for me when switching from the "IPv4 Debug STDIO" to the "IPv4 Debug" stack.

FerdinandEn, 2017-01-23:
I have reported this bug to Keil. It will be solved in version 7.4.0 of Middleware.
The reason is that `osTimerDef()` is part of compiled network library. My workaround is a modification of `cmsis_os_v1.c `
Change
```
osTimerId osTimerCreate (const osTimerDef_t *timer_def, os_timer_type type, void *argument) {

  if (timer_def == NULL) {
    return NULL;
  }
// return osTimerNew((osTimerFunc_t)timer_def->ptimer, type, argument, &timer_def->attr);
return osTimerNew((osTimerFunc_t)timer_def->ptimer, type, argument, NULL);
}
```

ReinhardKeil, 2017-02-13:
Middleware 7.4.0 is now available

closed 2017-02-13 by ReinhardKeil