ARM-software/CMSIS_5 #135: ARMC6/GNUC block causes errors in ARMC5

bulislaw, 2017-01-17:
Hi, I'm trying to update mbed-os CMSIS/RTX to the newest version from develop branch. This block is causing issues:
```
#if (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)) || \
     defined (__GNUC__)

extern __attribute__((weak)) uint32_t __os_thread_cb_start__;
extern __attribute__((weak)) uint32_t __os_thread_cb_end__;
extern __attribute__((weak)) uint32_t __os_timer_cb_start__;
extern __attribute__((weak)) uint32_t __os_timer_cb_end__;
extern __attribute__((weak)) uint32_t __os_evflags_cb_start__;
extern __attribute__((weak)) uint32_t __os_evflags_cb_end__;
extern __attribute__((weak)) uint32_t __os_mutex_cb_start__;
extern __attribute__((weak)) uint32_t __os_mutex_cb_end__;
extern __attribute__((weak)) uint32_t __os_semaphore_cb_start__;
extern __attribute__((weak)) uint32_t __os_semaphore_cb_end__;
extern __attribute__((weak)) uint32_t __os_mempool_cb_start__;
extern __attribute__((weak)) uint32_t __os_mempool_cb_end__;
extern __attribute__((weak)) uint32_t __os_msgqueue_cb_start__;
extern __attribute__((weak)) uint32_t __os_msgqueue_cb_end__;

__asm (".global os_cb_sections");

extern const uint32_t os_cb_sections[];

__attribute__((section(".rodata")))
const uint32_t os_cb_sections[] = {
  (uint32_t)&__os_thread_cb_start__,
  (uint32_t)&__os_thread_cb_end__,
  (uint32_t)&__os_timer_cb_start__,
  (uint32_t)&__os_timer_cb_end__,
  (uint32_t)&__os_evflags_cb_start__,
  (uint32_t)&__os_evflags_cb_end__,
  (uint32_t)&__os_mutex_cb_start__,
  (uint32_t)&__os_mutex_cb_end__,
  (uint32_t)&__os_semaphore_cb_start__,
  (uint32_t)&__os_semaphore_cb_end__,
  (uint32_t)&__os_mempool_cb_start__,
  (uint32_t)&__os_mempool_cb_end__,
  (uint32_t)&__os_msgqueue_cb_start__,
  (uint32_t)&__os_msgqueue_cb_end__
};

#endif
```
For ARM Compiler 5.06 it seems that ` || defined (__GNUC__)` makes the compiler to go inside the block and error out:
```
[Error] rtx_lib.c@482,0: #412: an "asm" declaration is not allowed here
```
line 482 is 
```
__asm (".global os_cb_sections");
```

RobertRostohar, 2017-01-17:
`__GNUC__ `is not defined for ARM Compiler 5 and the mentioned code should not be compiled in that case. This code has been verified with ARM Compiler 5.06 and it compiles without issues. 

Check if your environment defines the mentioned symbol somewhere.

0xc0170, 2017-01-17:
> __GNUC__is not defined for ARM Compiler 5 and the mentioned code should not be compiled in that case. This code has been verified with ARM Compiler 5.06 and it compiles without issues.

It is defined if ``--gnu`` flag is defined for ARMCC. Is that code above valid for this use case?
I had this problem previously when checking for only ``GNUC`` is not sufficient in some cases when using ARMCC with --gnu flag enabled.

RobertRostohar, 2017-01-18:
ARMCC5 provides GNU extensions when `--gnu` is used and also defines `__GNUC__ `. `asm `keyword should be supported in this mode but it seems that not the use case listed here.

Additional check for ARMCC5 has been added to exclude this.

bulislaw, 2017-01-18:
Thanks, that fixes it. Closing.

closed 2017-01-18 by bulislaw