ARM-software/CMSIS_5 #137: Vendor logo Icon/image

mzamaron, 2017-01-23:
I've raised an issue on the following GitHub, not the right.

My proposal is to have:
I would need to be able to specify the path to the file where we can found an icon. The goal is to be able to display it as a logo into a tool which manage CMSIS-Pack.�
Is it possible to add an element into the pdsc scheme to describe the file into the pack for this logo.

ReinhardKeil, 2017-02-07:
At which level is this icon file needed?  At the <package> element level (so that it applies to the whole pack, or for different elements. what is the icon size that you are proposing?

jkrech, 2017-02-08:
The intention for this logo is to represent the pack <vendor>, hence the tag should be located on the package level.

`<logo>filepath/filename.extension</logo>`

We need to agree on supported file formats and minimum dimensions.

JonatanAntoni, 2019-03-12:
@jkrech any update on this one?