ARM-software/CMSIS_5 #138: Keyword

mzamaron, 2017-01-23:
In CMSIS a couple of keywords are proposes:
 Board Support, CMSIS, Device, CMSIS Driver
 File System, Graphics, Network, USB for cClass attribute.
 CORE, DSP, RTOS for cGroup attribute.
 I would like to know if it's possible to deploy standard keywords across partners/vendors to easily match the conditions on external packs components for Group, Variant, bundle,..
 The list of keywords must be accessible on the Keil.com web site.

ReinhardKeil, 2017-02-07:
We did add further clarifications to https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DoxyGen/Pack/src/pack_creation.txt#63

This will go live in the next few days.  I agree with you that we need to do further work here.

ReinhardKeil, 2017-02-13:
released as part of CMSIS 5.0.1

closed 2017-02-13 by ReinhardKeil