ARM-software/CMSIS_5 #141: CMSIS __WEAK macros implementation for MDK compiler is wrong

IuriiB, 2017-02-06:
CMSIS __WEAK macros implementation for MDK (ARMCC) compiler is wrong.
CMSIS __WEAK macros is used __attribute__((weak)) which is not functional for MDK compiler - because in case when strong function, with the same name is declared somewhere in code, than compiler treats it wrong and throw an compilation error that there are two objects with the same names.

Per ARMCC documentation:
"Functions declared with __attribute__((weak)) and then defined without __attribute__((weak)) behave as weak functions. This is not the same behavior as the __weak keyword."

"Functions defined with __weak export their symbols weakly. A weakly defined function behaves like a normally defined function unless a nonweakly defined function of the same name is linked into the same image. If both a nonweakly defined function and a weakly defined function exist in the same image then all calls to the function resolve to call the nonweak function. If multiple weak definitions are available, the linker generates an error message, unless the linker option --muldefweak is used. In this case, the linker chooses one for use by all calls.

Functions declared with __weak and then defined without __weak behave as nonweak functions."

Using __weak for MDK (ARMCC) compiler solves the issue.

GuentherMartin, 2017-02-06:
Hello IuriiB,

I did some tests and noticed the differences between __weak and __attribute__((weak)) with functions.
__weak generates weak functions only when used in the function definition.
__attribute__((weak)) generates weak function in any case. If it used in function definition or in function declaration.

In my opinion it is better to use __attribute__((weak)) than to use __weak for ARMCC because it covers more used cases.

Maybe I have a wrong understanding.
Could you please give me an example how you use __WEAK in your code.

Best regards, Martin

IuriiB, 2017-02-06:
Hi Martin,

I have testFunc defined in mylib.c and prototype in mylib.h like the following weak function:
mylib.c: __WEAK void testFunc (void) { /* some code */ }
mylib.h: __WEAK void testFunc (void);

CMSIS provides the following implementation for __WEAK macro for MDK compiler:
#define __WEAK                    __attribute__((weak))

Then I add strong implementation of testFunc to my main_cm0p.c like:
void testFunc (void) { /* some code */ }

and the following error appears during project compiling:
ERROR: L6200E: Symbol testFunc multiply defined (by mylib.o and main_cm0p.o).

When I use "__weak" instead "__WEAK" then project compiles without errors.

Regards,
Iurii

GuentherMartin, 2017-02-07:
Hello Iurii,

if __weak is used instead of __attribute__((weak)) for macro __WEAK for ARMCC compiler than your code compiles without errors but your code is no longer portable to other compilers like GCC or ARMCLANG.

I would suggest to remove __WEAK from the function declaration in mylib.h. Than macro __WEAK can stay as it is and your code is portable to other compilers.

What was the intention to add __WEAK to the function declaration?

Best Regards, Martin

ReinhardKeil, 2017-02-07:
Are you using the same header file (mylib.h) in your strong implementation file (main_cm0p.c).  If so that the function is weak also in main_cm0p.c which would explain the problem.

You should remove the 'weak' definition in mylib.h

IuriiB, 2017-02-08:
Hi Martin and Reinhard,

1) I need weak function for fault handler implementation with a default behavior and possibility for the user to redefine it with other required behavior in strong function.
2) CMSIS __WEAK provides implementation for different compilers (GCC, MDK, IAR), but for MDK the implementation is not functional for the functions. Currently in my project I have another define which is compiler specific as replacement to CMSIS __WEAK. But I expect some fix in CMSIS for the MDLK compiler so I can use __WEAK macro for my project.

Regards,
Iurii

chentang16, 2017-02-10:
Hi Iurii,

I made further tests based on the non-modified version of cmsis_compiler.h and cmsis_armcc.h. In other word, the macro __WEAK is defined as __attribute__((weak))

Test 1): as you mentioned:
mylib.c: __WEAK void testFunc (void) { /* some code */ }
mylib.h: __WEAK void testFunc (void);
In this case both definition and declaration of testFunc are weak. 
And if I add #include "mylib.h" into my main.c and "redefine" nonweak void testFunc (void){ }; in main.c, then I got the error you mentioned "ERROR: L6200E: Symbol testFunc multiply defined" 
That is because of the reason as Reinhard explained before. In this case testFunc is weak in main.c as well. 
This compilation error can be resolved by adding --muldefweak to linker. Then you will see the warning "L6439W: Multiply defined Global Symbol testFunc defined in .text(mylib.o) rejected in favor of Symbol defined in .text(blinky.o)."

Test 2): as Reinhard suggested:
mylib.c: __WEAK void testFunc (void) { /* some code */ }
mylib.h: void testFunc (void);
In this case the definition of testFunc is weak, but the declaration of testFunc is nonweak. 
And I do the same as test 1 (add #include "mylib.h" into my main.c and "redefine" nonweak void testFunc (void){ }; in main.c,). Then I got no compilation error without --muldefweak. According to callgraph and map file, the nonweak testFunc definition in main.c is actually used, because testFunc in mylib.c is still weak in this case.

Test 3):
mylib.c: __WEAK void testFunc (void) { /* some code */ }
mylib.h: __WEAK void testFunc (void);
But in this test, I didn't add #include "mylib.h" into my main.c, but still "redefine" nonweak void testFunc (void){ }; in main.c. In this case still no compilation error without --muldefweak. According to callgraph and map file, the nonweak testFunc definition in main.c is actually used, because testFunc in mylib.c and mylib.c is still weak in this case.

Thus, I don't see the reason why the current macro __WEAK implementation in CMSIS should be "fixed".

IuriiB, 2017-02-14:
Hi Chen Tang,

It looks like your test #2 should work for my case. 

Thank you,
Iurii

ReinhardKeil, 2017-02-27:
closing this as there is no further activity

closed 2017-02-27 by ReinhardKeil