ARM-software/CMSIS_5 #147: Segmentation fault when description field > +-570 characters

JakkoVerhallen, 2017-02-15:
Hi,

I'm experiencing segmentation faults when the description of a register field is too long.
It should be quite easy to reproduce.

I'm using SVDConv 4.4.4 on Linux RH6.

Thanks!
Jakko

JonatanAntoni, 2017-03-03:
@JakkoVerhallen may I ask you to review the latest version of SVDConv for Linux and close this issue? Thx.

JakkoVerhallen, 2017-03-05:
Hi Jonatan,

Thanks for your email.
I have checked the latest version, and I don’t need to remove long descriptions anymore, so it seems to be fixed!

Is there any other action I should take? I’m quite new to github.

Thanks!
Jakko

Legal Disclaimer: This e-mail communication (and any attachment/s) is confidential and contains proprietary information, some or all of which may be legally privileged. It is intended solely for the use of the individual or entity to which it is addressed. Access to this email by anyone else is unauthorized. If you are not the intended recipient, any disclosure, copying, distribution or any action taken or omitted to be taken in reliance on it, is prohibited and may be unlawful.

Please consider the environment before printing this e-mail

thorstendb-ARM, 2017-03-06:
Hi,

the Seg.Fault should be fixed in SVDConv Linux v3.3.3

closed 2017-03-06 by JonatanAntoni