ARM-software/CMSIS_5 #149: cmsis_iar file missing in the tree

0xc0170, 2017-02-16:
Include is here : https://github.com/ARM-software/CMSIS_5/blob/c5e87e54ffe69be548823be030b0d2169eb2b707/CMSIS/Core/Include/cmsis_compiler.h#L66

But I can not locate it in the tree, thus some definitions for IAR are missing.

jkrech, 2017-02-16:
The file cmsis_iar.h is shipped by IAR as part of their tool-chain.
If you have EWARM installed you should be able to locate it there or if you are not using IAR your tool-chain should ignore this section of the file completely as you must not have __ICCARM__ being defined.

0xc0170, 2017-02-16:
Thanks ! I was able to find the file, however I am still hitting the same problem that I suspected due to missing header file. I explored that cmsis_iar.h does not contain any definitions for this error:

```
  #error "Unknown ARM Architecture!"
   ^
"rtos\rtx2\TARGET_CORTEX_M\core_cm.h",56  Fatal error[Pe035]: #error directive: "Unknown ARM Architecture!"
```

GCC defines them, for armcc there is a translation between their symbols and this __ARM_ARCH. What about for IAR? I noticed that this __ARM_ARCH macro is defined in one of .ewp files that does not look right. I am not using this .ewp but only sources from this repository.

0xc0170, 2017-02-16:
Should this be part of cmsis_compiler.h for IAR specific part:

```
#if (__CORE__ == __ARM6M__) || (__CORE__ == __ARM6SM__)
#define __ARM_ARCH_6M__ 1
#elif (__CORE__ == __ARM7M__)
#define __ARM_ARCH_7M__ 1
#elif (__CORE__ == __ARM7EM__)
#define __ARM_ARCH_7EM__ 1
#endif
```

Then those symbols __ARM_ARCH_xx__ could be removed from .ewp project files, the sources should compile.

I can't find what version of IAR was used for those project files, I got 7.70.1 and receive a warning when I open a project, that means it was generated by older version of IAR (broken options detected warning).

0xc0170, 2017-02-16:
Once I defined __ARM_ARCH_ macros for IAR as above, I progressed with errors in our code base, and hit this one:

```
rtx_msgqueue.c",111  Error[Ta117]: IT-block ends prematurely, or there is a label within the IT-block
Internal error: [AsmLine - OgAsm]: Error[425]: Conditional instruction is only allowed inside IT block
        UXTBNE   R1,R1
Fatal error detected, aborting.
```

It looks to me that ``atomic_wr8`` is causing problems for IAR asm.

0xc0170, 2017-02-16:
```
Error[Li006]: duplicate definitions for "osRtxUserSVC"; in "rtx_lib.o"
          , and "BUILD\K64F\IAR\mbed-os\rtos\rtx2\T
          ARGET_CORTEX_M\svc_user.o"
[ERROR] Error[Li006]: duplicate definitions for "osRtxUserSVC"; in "rtx_lib.o"
          , and "BUILD\K64F\IAR\mbed-os\rtos\rtx2\T
          ARGET_CORTEX_M\svc_user.o"
```

Once we resolve these problems ( I commented them out at  the moment), my blinky app compiles ! Please review this errors

seems like osRtxUserSVC is not getting proper weakly linkage in this case.

0xc0170, 2017-02-16:
I sent 2 pull requests fixing problems I described here. The last one that I would like to fix is ``atomic_wr8``.

0xc0170, 2017-02-20:
Any suggestions? I would like to progress regarding these 2:

I downloaded latest IAR beta, had some another internal error that did not allow me to continue, however did not spot this error anymore. A question is what is special for IAR , why I could not compile with 7.7 IAR version?

>rtx_msgqueue.c",111  Error[Ta117]: IT-block ends prematurely, or there is a label within the IT-block
Internal error: [AsmLine - OgAsm]: Error[425]: Conditional instruction is only allowed inside IT block
        UXTBNE   R1,R1
Fatal error detected, aborting.

and this duplicate symbol. Is svc_user.c intended to be in the application ? It looks like a template and defines a strong symbol. I patched it differently in the referenced PR above.

> Error[Li006]: duplicate definitions for "osRtxUserSVC"; in "rtx_lib.o"
          , and "BUILD\K64F\IAR\mbed-os\rtos\rtx2\T
          ARGET_CORTEX_M\svc_user.o"
[ERROR] Error[Li006]: duplicate definitions for "osRtxUserSVC"; in "rtx_lib.o"
          , and "BUILD\K64F\IAR\mbed-os\rtos\rtx2\T
          ARGET_CORTEX_M\svc_user.o"

0xc0170, 2017-03-01:
> I downloaded latest IAR beta, had some another internal error that did not allow me to continue, however did not spot this error anymore. A question is what is special for IAR , why I could not compile with 7.7 IAR version?

    rtx_msgqueue.c",111 Error[Ta117]: IT-block ends prematurely, or there is a label within the IT-block
    Internal error: [AsmLine - OgAsm]: Error[425]: Conditional instruction is only allowed inside IT block
    UXTBNE R1,R1
    Fatal error detected, aborting.

I switched to the latest stable IAR version (7.8) and this problem is not there.

JonatanAntoni, 2017-03-02:
With IAR EWARM 7.80.4 and PR #150 this issue should be fixed.

JonatanAntoni, 2017-03-13:
With IAR EWARM 8.10 the missing defines are announced to be part of `cmsis_iar.h` shipped by IAR.

JonatanAntoni, 2017-04-25:
We now introduced the workaround for backward compatibility reasons into cmsis_compiler.h.

0xc0170, 2017-04-25:
I am closing this, as now the sources should be allowed to compile with IAR 7.80. Thanks for the help @JonatanAntoni

closed 2017-04-25 by 0xc0170