ARM-software/CMSIS_5 #152: IAR: ALIGNED not supported warning in the header file might result in lot of warnings

0xc0170, 2017-02-16:
The warning is fine to warn a user that something is not supported by a compiler but the placement of this warning might be counterproductive - in a header file. It's currently in a header that is being included everywhere as it's part of CMSIS. Therefore when a user compiles their code, it might end up with 100x warnings (all are the same, for each file that depends on CMSIS). It creates lot of noise, that will result in removing this warning after few compilations :-)

I would suggest to have one code file  that would warn about this - warning: this feature of cmsis is not supported by this compiler. Or remove that warning and document it what is supported and what not.

ReinhardKeil, 2017-02-27:
I think you should be able to use a compiler directive that effectively inserts an #define __ALIGNED. In this way you can avoid the issue.

P.S. IAR is working on a solution to that problem.

0xc0170, 2017-02-28:
I am missing how this was resolved (DONE label) ? The warning was removed? IAR new version needed or?

JonatanAntoni, 2017-03-02:
IAR seems to have solved the issue. With EW-ARM 7.80.4 IAR ships a `cmsis_iar.h` header file containing the required macro definition as 
```
#ifndef __ALIGNED
#define __ALIGNED(x) __attribute__((aligned(x)))
#endif
```

0xc0170, 2017-03-02:
I'll close this issue, thanks!

closed 2017-03-02 by 0xc0170