ARM-software/CMSIS_5 #153: Incorrect code comments in DSP library

rogersnm, 2017-02-27:
The code comments in the DSP library are not correct. This is a hangover from a bug in the previous versions of the DSP library (see #148 for details).

From arm_math.h:
```
#if defined(ARM_MATH_CM7)
  #include "core_cm7.h"
  #define ARM_MATH_DSP
#elif defined (ARM_MATH_CM4)
  #include "core_cm4.h"
  #define ARM_MATH_DSP
#elif defined (ARM_MATH_CM3)
  #include "core_cm3.h"
#elif defined (ARM_MATH_CM0)
  #include "core_cm0.h"
  #define ARM_MATH_CM0_FAMILY
#elif defined (ARM_MATH_CM0PLUS)
  #include "core_cm0plus.h"
  #define ARM_MATH_CM0_FAMILY
#elif defined (ARM_MATH_ARMV8MBL)
  #include "core_armv8mbl.h"
  #define ARM_MATH_CM0_FAMILY
#elif defined (ARM_MATH_ARMV8MML)
  #include "core_armv8mml.h"
  #if (defined (__DSP_PRESENT) && (__DSP_PRESENT == 1))
    #define ARM_MATH_DSP
  #endif
#else
  #error "Define according the used Cortex core ARM_MATH_CM7, ARM_MATH_CM4, ARM_MATH_CM3, ARM_MATH_CM0PLUS, ARM_MATH_CM0, ARM_MATH_ARMV8MBL, ARM_MATH_ARMV8MML"
#endif
```

We can see that ARM_MATH_DSP will be defined for the CM7, CM4 and possibly the ARMV8. 

In the DSP library I counted 168 occurrences over 168 files of the code:

```
#if defined (ARM_MATH_DSP)

/* Run the below code for Cortex-M4 and Cortex-M3 */
```

(eg. https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/BasicMathFunctions/arm_add_q15.c)

As ARM_MATH_DSP is not defined for the CM3, the code comment is incorrect and should read something more along the lines of:

```
#if defined (ARM_MATH_DSP)

/* Run the below code for Cortex-M7, Cortex-M4 and some ARMV8 platforms  */
```

This effects the following files:
```
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_abs_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_add_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_add_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_add_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_add_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_dot_prod_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_mult_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_mult_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_mult_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_mult_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/arm_negate_f32.c (1 hit)
[... 125 lines cut ...]
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_copy_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_copy_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_fill_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_fill_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_fill_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_fill_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_float_to_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_float_to_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_float_to_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q15_to_float.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q15_to_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q15_to_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q31_to_float.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q31_to_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q31_to_q7.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q7_to_float.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q7_to_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_q7_to_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_cfft_radix4_q15.c (2 hits)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_dct4_f32.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_dct4_q15.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_dct4_q31.c (1 hit)
/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q15.c (2 hits)
```

pallaviboreddy, 2020-01-22:
when i executed "arm_convolution_example_f32.c" i get the following errors
in function "main"
undefined reference to arm_fill_f32
undefined reference to arm_copy_f32
undefined reference to arm_cfft_radix4_init_f32
undefined reference to arm_cfft_radix_f32
undefined reference to arm_cmplx_mult_cmplx_f32
undefined reference to arm_snr_f32
collect2: error: ld returned 1 exit status
please suggest to overcome these errors

christophe0606, 2020-01-22:
@pallaviboreddy Can you open a new github issue ?
It looks like your problem is quite different from the one discussed in this issue.

christophe0606, 2020-01-23:
Regarding the original issue : it no more applies to the code as it is today. So, I close this issue.

closed 2020-01-23 by christophe0606