ARM-software/CMSIS_5 #155: CMSIS-SVD validator should be open source

odinthenerd, 2017-02-28:
It would be much easier for the community to make pull requests adding validator checks than filing issues and waiting for ARM. Please open source it.

We would gladly implement [this](https://github.com/ARM-software/CMSIS_5/issues/154) and other checks.

ilg-ul, 2017-02-28:
an open source SVD validator would be great, and a portable written one would be even greater.

the current preferred solution would be a [npm](https://www.npmjs.com) package ([node.js](https://nodejs.org/en/) based).

the _yesterday_ preferred solution was Python, but given the mess with different versions, and the traction that npm has nowadays, Python is already somehow deprecated.

as for performances... I doubt they justify having this program written in C++.

odinthenerd, 2017-02-28:
I would of course prefer modern C++ but I don't really care, as long as it is open is can be in haskell for all I care

ilg-ul, 2017-02-28:
> prefer modern C++

yes, sure, but then you also need a portable build environment (will it work on windows/linux/macos? will it be autotools? cmake? on windows, will it be mingw-w64 or microsoft?), then you also need to have **exactly** the same libraries, and so on.

not really worth the effort, imho.

xml processing is a piece of cake for modern interpreted languages. for node.js you install only the interpreter, then all dependencies for the application are automatically pulled in, nothing else to do.

David-Garcia-Polo, 2017-03-01:
I still prefer the "yesterday" solution in python, but one way or the other will work.

Nevertheless, as long as for example the pack installer in uVision uses the installed SVDConv there is no solution for pack installation in that environment.

JonatanAntoni, 2017-03-21:
We have discussed and decided to not open source SVDConv in the foreseeable future.

JonatanAntoni, 2017-04-03:
Closing this issue due to no further feedback or questions raised.

closed 2017-04-03 by JonatanAntoni