ARM-software/CMSIS_5 #156: CMSIS-SVD should include better FIFO descriptors

odinthenerd, 2017-02-28:
The vast majority of hardware FIFOs follow the pattern of a fixed address register to which one can write until a bit in another register changes state (full flag or something). Or in the case of an input FIFO one can read until an empty flag is set. 

I CMSIS-SVD would provide an interface with which the FIFO register is associated with the corresponding full/empty flag (as well at the sate, some are inverse logic) it would allow us to generate C++ ranges out of this data which would be a great help to the user.

JonatanAntoni, 2017-02-28:
As far as understand your intention, you aim to transpose the C solution (pseudo code)
```C
while(!FIFO_EMPTY) {
   byte = FIFO_RX;
   // use byte for processing
}
```

into a C++11 solution like
```C++
for(auto byte: FIFO) {
   // use byte for processing
}
```

Please come up with a solution, i.e. enhanced version of the [SVD schema](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Utilities/CMSIS-SVD.xsd). Please keep in mind that the SVD is a static view of the system, not a dynamic one. Thus the dynamics of a real FIFO (which might be very different between implementations) have to be negligible.

odinthenerd, 2017-02-28:
yes pretty much, although this hits the nail on the head better:
```C++
void onUartRx(){
    copy(uart0rx | byteToAscii, pushTo(uartQueue));
}
void onUsbSof(){
    copy(popFrom(uartQueue) , spi1tx);  //out range v3 can fail so no overflow
}
```
the many chips I have could all be abstracted using a "fifo bitfield" and a "status bitfield" as well as a "status value" which can be compared to that bitfield in order to determine if it is full or empty. I'm sure there is some obscure chip that does not match that but if the description is not manditory thats is probably not a problem.

JonatanAntoni, 2017-03-03:
Sounds interesting, but exceeds our current scope. You might make use of the "Vendor Extensions" in SVD for a separate solution for now.

JonatanAntoni, 2017-03-13:
I close this issue for now.

closed 2017-03-13 by JonatanAntoni