ARM-software/CMSIS_5 #157: maschiene readable datasheets

odinthenerd, 2017-02-28:
although this is a bit off topic here I think its relevant. 

CMSIS-SVD is essentially a machine readable hardware description, however it only describes a small part of the functionality of the chip. It was discussed at the last [emBO++](embo.io) as well as the following SG14 meeting that a more through machine readable datasheet is key to raising the level of abstracting and the portability and quality of drivers between chips as well as static validation tools. 

I can't imagine that the vendor datasheets are fully written by hand so most of this data must already exist in machine readable form. How can we best convince vendors to provide this information? Can ARM expand CMSIS to include more of this?

ilg-ul, 2017-02-28:
I use part of the SVD definitions to create the peripheral registers in [GNU ARM Eclipse QEMU](http://gnuarmeclipse.github.io/qemu/).

I think there are many issues with the current SVDs, the information might be there, but some of the definitions are odd, inconsistent between families, and each vendor has its own understanding of the SVD specs, which makes real life use of the SVDs very difficult.

until a very detailed representation of the datasheet (which would be great) will be available, perhaps a more strict approach would be useful.

David-Garcia-Polo, 2017-02-28:
I support the request. For generating header files for our devices I am still dependent on a very old version of SVDConv. would like to be able to fix it myself.

ilg-ul, 2017-02-28:
> For generating header files ... a very old version of SVDConv

for generating files from SVDs, I did some experiments and things are quite easy with some [Liquid](http://shopify.github.io/liquid/) templates, in node.js:

https://github.com/xcdl/xcdl-js/tree/master/lib/xcdl/svd

in GNU ARM Eclipse QEMU I need some JSON peripheral representations, which are generated from SVDs, then patched with missing information, and also lots of source files are generated. you can take a look at https://github.com/gnuarmeclipse/qemu/tree/gnuarmeclipse-dev/gnuarmeclipse/devices/support

JonatanAntoni, 2017-02-28:
David, can you please provide an issue or bug report about your trouble with the current version of [SVDconv](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Utilities/Win32/SVDConv.exe)?

odinthenerd, 2017-02-28:
https://github.com/kvasir-io/HardwareDescriptionFileGenerator is what we are working on to deal with the discrepancies between vendors. We would value your input @ilg-ul.

At the end of the day I don't care what format its in, if vendors would just dump buggy excel sheets on the net warts and all with a proper license it would be far better than nothing. What people like http://xpcc.io/ or https://github.com/kvasir-io or others will do with the data will also be far better than what vendors are publishing in code now. 

The industry is learning that serious vendor lock in is a liability and essentially high interest tech debt, I think there is a first mover advantage on the part of the vendors.

ilg-ul, 2017-02-28:
> HardwareDescriptionFileGenerator ... value your input

sure, but for now it is empty, ping me when more content.

> xpcc.io ... kwasir-io ...

yes, interesting solutions. when xPack (my package management tools) will be functional, perhaps we can make some packages of them and use together with µOS++ / CMSIS++.

odinthenerd, 2017-02-28:
> when xPack (my package management tools) will be functional, perhaps we can make some packages of them and use together with µOS++ / CMSIS++.

sounds interesting

> sure, but for now it is empty, ping me when more content.

sorry should have mentioned, look at the development branch, its a git flow style and we have not released version one yet (coming week will be 1.0 I think) so nothing in master

ilg-ul, 2017-02-28:
> look at the development branch

I did, please check the issue I registered.

David-Garcia-Polo, 2017-03-01:
@JonatanAntoni  The last issue was closed on Friday (https://github.com/ARM-software/CMSIS_5/issues/144) I am opening a new one today regarding the treatment of derivedPeripheral structs.

JonatanAntoni, 2017-03-21:
@odinthenerd we are continuously working on enhancements here. Instead of pushing more device details to the plain "old" SVD files (which were initially designed for the system viewer) we think about enhancing the pack description. But pushing the pack XSD further is only half the way. I cannot promise that all the pack maintainers will provide all the information you are looking for.

JonatanAntoni, 2017-08-29:
Hi @odinthenerd,

we are pushing forward standardization as best as we can, e.g. take a look to CMSIS-Zone. But as you can imagine standardization is a tough business. We have to clearly highlight the advantages and the business value in order to convince our partners.

Do you have further thoughts to be discussed here? Or do you mind to close the issue?

Cheers,
Jonatan