ARM-software/CMSIS_5 #158: documentation fo clock trees should be part of CMSIS-SVD

odinthenerd, 2017-02-28:
changing the master clock frequency and propagating all those changes through all peripheral configurations is way way harder than it should be. This is also often a hurdle to get things running. 

Essentially every vendor must have this data and STM even offers tooling to handle clock trees. On the other side many libs handle baud calculation etc. dynamically at runtime which is just not efficient at all. Modern languages with built in code generation could easily handle this at compile time (C++ constexpr or TMP, rust language extensions etc.) if we had the clock tree info. 

I can't imagine it is in ARMs interest to allow a few vendors and third parties to play the vendor lock in game with this stuff, it undermines ARMs unique sales pitch of compatibility.

ilg-ul, 2017-02-28:
that would be great, but for now even simple things like a complete list of interrupts with official CMSIS handler routine names is not available. this would allow to generate the vectors automatically, and do not rely on vendor provided assembly files.

odinthenerd, 2017-02-28:
some info is in there, using C++ we did a proof a concept a while ago where we compile time generated the interrupt vector table with the correct pointers based on a list of input classes. The user just needed to place a macro in main which included all the classes which need interrupts as parameters. The macro expanded to (calls to) all the start up code plus the generated vector table and some libc stubs so that the user need not know how any of that stuff works or include any other source files. 

That solution had the advantage of not needing to know the official names of the ISRs and made it easy for the user to move the vector table to RAM by just dropping in a tag. We may pursue this again at some point.

ilg-ul, 2017-02-28:
> some info is in there

yes, some info is in there, but, if I remember right, when I did some tests, I found a case when some interrupts were missing.

maybe things improved, but for this mechanism to be usable, we need guarantees (or at least specifications claiming this), not guesses.

and the SVD validator must check this.

JonatanAntoni, 2017-03-01:
May I ask you to come up with a proposal on how you would like to have the clock information listed in an SVD file?

ReinhardKeil, 2017-03-02:
What would be the use case of this information. It should be noted that clock trees of devices can be extremely complex and each silicon vendor has a different schema.

JonatanAntoni, 2017-03-08:
As Reinhard already noted are the clock trees vendor specific. The clock tree capabilities of different devices may vary in a large range; making it very difficult to describe them in a generic way. Please take note that vendors are coming up with individual solutions for their chips, i.e. STM32CubeMX.

At this point in time this issue is not in scope for CMSIS. We are looking forward to see which solutions the vendors and the community are going publish. So please feel free to start off with a separate solution loosely coupled with CMSIS. We appreciate if you report back on your progress.

JonatanAntoni, 2017-03-21:
Closing due to inactivity.

closed 2017-03-21 by JonatanAntoni