ARM-software/CMSIS_5 #162: CMSIS RTOS2: Service calls fail under GCC with optimization enabled

arvinf, 2017-03-13:
Compiling with any optimization flags  causes a lock up when any service calls are made. I've tracked the issue down to [SVC_Handler](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/GCC/irq_cm0.S#L52). Position of SVC Instruction in stack changes between non-optimized and optimized builds.

RTX should not rely on hardcoding position parameters to pass them to service call. Important values can be passed using registers.

Tested with `gcc version 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496] (GNU Tools for ARM Embedded Processors)` on Cortex-M0 (STM32F070)

RobertRostohar, 2017-03-14:
Can you provide a reproducer?

RTOS2 has been tested with the mentioned gcc version and compiles and runs with all optimization levels. There is a different issue when no optimization is used (issue #118).

> Important values can be passed using registers.

No, since a higher priority interrupt can overwrite the registers. Parameters need to be passed over stack.

arvinf, 2017-03-15:
Upon further investigation the problem is due to calling `osKernelGetTickCount` after `osKernelInitialize` but before `osKernelStart`. In the documentation this is not specifically prohibited, only that  `osKernelGetTickCount` should not be called inside an ISR.

Calling osKernelGetTickCount before osKernelStart means `sp` points to `msp` but the SVC call expects it to point to `psp` therefore incorrect stack is retrieved. When optimization is disabled, it just so happens that the incorrect stack points to a memory area that causes the subsequent SVC# check to correctly read as zero and the call succeed.

Documentation should be corrected to reflect that osKernelStart must be called before any SVC call. 

In this case osKernelGetTickCount was called by STM32F HAL init routines.

RobertRostohar, 2017-03-15:
Good to hear that you have found the real reason.

The documentation needs to be updated to clearly state which functions can be called before osKernelInitialize/osKernelStart.

Functions allowed to be called before osKernelInitialize:
- osKernelGetInfo
- osKernelGetState

Functions allowed to be called after osKernelInitialize and before osKernelStart:
- osKernelGetInfo
- osKernelGetState
- osXxxNew (all object creation functions)

closed 2017-03-15 by arvinf