ARM-software/CMSIS_5 #164: RTOS2: msgqueueu error with IAR 7.8 for cortex-m0 targets

0xc0170, 2017-03-17:
The code file msgqueue uses __CLZ: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_msgqueue.c#L258

Code snippet:

```
  if ((__CLZ(msg_count) + __CLZ(block_size)) < 32) {
    EvrRtxMessageQueueError(NULL, osErrorParameter);
    return NULL;
}
```

This fails to build with IRA 7.8, for any cortex-m0 device, the error: ``Error[Pe020]: identifier "__cmsis_iar_clz" is undefined`` . I could not locate where ``__cmsis_iar_clz`` is defined .

From ``cmsis_iar`` header file, this is redefinition of ``__CZL``:

```
#if (__CORE__ == __ARM6M__)
  /* Avoid clash between intrinsics.h and arm_math.h when compiling for Cortex-M0. */
  #define __CLZ __cmsis_iar_clz
  #define __SSAT __cmsis_iar_ssat
#endif
```

I searched for ``__cmsis_iar_clz`` in IAR library sources, no luck.

Has anyone being able to build RTOS2 with IAR 7.8 or around ?

ReinhardKeil, 2017-03-20:
We will replace CLZ on Cortex-M0/M0+/M23 with an alternative code.

0xc0170, 2017-04-06:
bump for the update

c1728p9, 2017-04-19:
As a workaround the implementation in arm_math.h can be used. I created a PR for this in mbed-os here https://github.com/ARMmbed/mbed-os/pull/4198.

JonatanAntoni, 2017-08-29:
Hi @0xc0170,

this issue should be resolved. May I ask you to close this issue if you do not have any further concerns, please?

Cheers,
Jonatan

JonatanAntoni, 2017-10-17:
Closing this issue. Please feel free to reopen if you have further questions.

closed 2017-10-17 by JonatanAntoni