ARM-software/CMSIS_5 #165: DSP: arm_conv_partial_f32 has bug

xizhizhang, 2017-03-21:
It seems there is a bug in arm_conv_partial_f32.c. 

Here is a Cortex-M4F test I have run on Keil's Simulation for this issue:
https://gist.github.com/xizhizhang/26e0be9b23db2540d7a72c4c61f25d11

We can see that in test1, arm_conv_partial_f32 is OK. But in test2 arm_conv_partial_f32 got different result with arm_conv_f32.

I am debugging on it. And I will pull a request if I can find out the bug.

xizhizhang, 2017-03-21:
In the algorithm's stage2, if we enter line 266, we get a px = pIn1 + firstIndex - srcBLen + 1:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_conv_partial_f32.c#L266

but, in line 423, we always update px as: px = pIn1 + count
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_conv_partial_f32.c#L423

this update should also check whether  firstIndex - srcBLen + 1 > 0

And six files have the issue:

- arm_conv_partial_f32.c
- arm_conv_partial_fast_q15.c
- arm_conv_partial_fast_q31.c
- arm_conv_partial_q15.c
- arm_conv_partial_q31.c
- arm_conv_partial_q7.c

llefaucheur, 2017-11-02:
Hi, thank you very much for the detailed explanation. Indeed 6 files are impacted,
For each file, the bug occurs three times. In _f32.c it occurs at #423, #476 and #513.
Lesson's learnt: never write such long source code without splitting it in subroutines or macros.
Regards, Laurent.

closed 2019-05-29 by christophe0606