ARM-software/CMSIS_5 #167: RTX primitives object storage sizes

bulislaw, 2017-03-23:
For non-dynamic allocation, memory can be statically provided using `os*Attr_t` structures. While porting mbed OS to RTX2 I exposed internals of RTX and using `rtx_lib.h` directly declared RTX object eg `os_mutex_t    _obj_mem;` that works fine, but it breaks abstraction layers and reduces portability. The other way would be to count how much memory is needed and declare something like `int32_t _mutex_data[4];` but it doesn't seem to be good solution either.
Do you guys think exposing sizes in `cmsis_os2.h` would be a good idea? We could do it in mbed OS, but I think the same issue would affect other users, if they don't want to have pools of memory statically declared at compile time.
https://github.com/ARMmbed/mbed-os/pull/4011

RobertRostohar, 2017-03-24:
> I exposed internals of RTX and using  rtx_lib.h 

You should not do that.

`cmsis_o2.h` provides information that is portable.

Implementation specific information (example: control block sizes) should be provided by the implementer.

For example RTX provides the header `rtx_os.h` with all RTX specifics including control block sizes (osRtxMutexCbSize, ...)

kjbracey-arm, 2017-03-24:
I see that you can't expose sizes in cmsis_os2.h itself - code written using that and dynamic allocation is fully portable, even as a binary library, which is a great achievement.

But in the event that a user is wanting to statically allocate objects, and as such they are accepting that their code will only be compile-time portable between different implementations, shouldn't there be standard naming for a header file defining "current implementation's objects"?

That would gives you two viable models:

    #include "cmsis_os2.h"
    // use dynamic allocation - be totally binary portable

and

    #include "cmsis_os2.h"
    #include "cmsis_os2_storage.h"
    // use static allocation appropriate to current implementation -
    // be compile-time portable

At the minute, rather than the second option, all the code has to do

    #include "cmsis_os2.h"
    #include "rtx_os.h"
    // Use static allocation, and support RTX only. Abstraction not serving its purpose.

RTX's cmsis_os2_storage.h would be basically

    #include "rtx_os.h"
    typedef os_mutex_t osMutexCb_t;
    // etc..

(I prefer using real types like osMutexCb_t to mere size numbers, as that also ensures precise alignment - neither excessive nor inadequate).

Of course, users could just define something like "cmsis_os2_storage.h" themselves without official approval, isolating their implementation-dependence into one file. Maybe if you don't want to encourage implementation-dependence, I can see that you wouldn't want to standardise such a header, to encourage them to use dynamic allocation. But given that there are already hooks all the way through the API to let people provide storage, it would seem reasonable to have a portable way to size that storage.

Otherwise the whole situation is a bit like having a portable fprints in "stdio.h", but no stderr - users having to include "stdio_macos.h" or "stdio_windows.h" to get either stderr_macos or stderr_windows.

BTW is there a run-time mechanism to get the object size, for the third possible approach of totally-portable users getting object size at run-time, and using their own allocator?

JonatanAntoni, 2017-09-01:
Hi all,

just to revive this discussion. We are going to standardize the object sized used in RTX (see [rtx_os.h](https://github.com/ARM-software/CMSIS_5/blob/401391fc370397eb7eef4cd9d8f49e0f1d80e8b3/CMSIS/RTOS2/RTX/Include/rtx_os.h#L356).

For the time being you can directly access rtx_os.h but for the future hopefully we can archive a more implementation agnostic solution.

Cheers,
Jonatan

embeddedteam103, 2018-12-17:
To revive discussion on this issue:
We have wrapped the `threadx` os with `cmsis` abstraction layer and therefore have different implementation.
The way we do this, among other things is replace the `mbed_rtos_storage.h` file with typedefs of `threadx`.
We ban dynamic allocation so the issue never came up.
I do have something to add tough.
I think adding the sizes as part of the public API as done in `RTX` is the right approach, an implementation with too large sizes that wants to use `cmsis` will have to add wrappers anyway - and in those wrappers - the size will be fixed.

The design question I see here is - is the ability to dynamically allocate `cmsis` objects in a portable way part of the API?
If the answer is yes, I don't see any other way then claiming the object sizes in the `cmsis` layer.

embeddedteam103, 2018-12-18:
I can start a PR of adding the sizes if someone official can tell me it will be accepted as the solution for this issue (I don't want to waste work time).