ARM-software/CMSIS_5 #168: two issues in arm_conv* files

xizhizhang, 2017-03-24:
Hi：
There are seems have two issues in some arm_conv* files.

first one is the comments, 

for example, at line 263 in arm_conv_f32:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_conv_f32.c#L263

x[srcALen-srcBLen-2] should be x[srcALen-srcBLen] , because stage2 have (srcALen - (srcBLen - 1u)) items.
so, this comment line should be:
"sum = =x[srcALen-srcBLen] * y[srcBLen-1] + x[srcALen-srcBLen+1] * y[srcBLen-2] +...+ x[srcALen-1] * y[0]"

Second is also about index, but is in code.
for example, at line 297 in arm_conv_f32:
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Source/FilteringFunctions/arm_conv_f32.c#L297

`x0 = *(px++);`
should be
`x0 = *px++;`

We should calculate convolution beginning at index 0, not 1. or we would got some errors

and here is my simple test case for this issue:
https://gist.github.com/xizhizhang/342841f7614c2906d9ee241bc72e3f5e

I copied arm_conv_f32's simply implementation, and renamed it as arm_conv_f32_simple in above case.
I think it is a convenient and intuitive case that shows arm_conv_f32 have a error.

Thanks.

xizhizhang, 2017-03-24:
for issue 1, there are:

- arm_conv_q15.c
- arm_conv_partial_q15.c
- arm_conv_f32.c
- arm_conv_fast_q31.c
- arm_conv_partial_q7.c
- arm_conv_q31.c
- arm_correlate_fast_q15.c
- arm_correlate_fast_q31.c
- arm_conv_partial_fast_q15.c
- arm_correlate_q15.c
- arm_correlate_q31.c
- arm_conv_partial_fast_q31.c
- arm_conv_fast_q15.c
- arm_correlate_q7.c
- arm_conv_partial_f32.c
- arm_conv_q7.c
- arm_conv_partial_q31.c
- arm_correlate_f32.c

for issue 2, there are:

- arm_conv_q7.c
- arm_conv_q31.c
- arm_conv_partial_q7.c
- arm_conv_partial_q31.c
- arm_conv_partial_fast_q31.c
- arm_conv_partial_f32.c
- arm_conv_fast_q31.c
- arm_conv_f32.c

xizhizhang, 2017-03-24:
https://github.com/ARM-software/CMSIS_5/pull/166/commits/c0b44f948ae4545bb71d3b202746381d60f941fc
is the pr for issue 2

christophe0606, 2020-01-22:
The code for arm_conv and arm_correlate has changed so much since this issue was initially reported than I cannot anymore know if it still applies.

So, I'll close the issue.

closed 2020-01-22 by christophe0606