ARM-software/CMSIS_5 #169: SVDConv utilities are out-of-sync with CMSIS-SVD schema version 1.3.3

eperie, 2017-03-28:
Version impacted: 729bd0ba738a4c5e695fe6a23d2a0d787e7b73a1, and likely CMSIS 5.0.1 - not verified yet.
Steps to reproduce:

1. Add the following line after line 62 of ARM_Example.svd:
`<vendorSystickConfig>false</vendorSystickConfig`
becomes:
`<vendorSystickConfig>false</vendorSystickConfig`
`<deviceNumInterrupts>15</deviceNumInterrupts>`
2. Validate modified ARM_Example.svd against CMSIS-SVD.xsd using xmllint:
`xmllint --schema CMSIS-SVD.xsd ARM_Example.svd`
  modified ARM_Example.svd does validate:
_ARM_Example.svd validates_
3. Attempt to generate header file from modified  ARM_Example.svd:
`Linux-gcc-4.8.3/SVDConv ARM_Example.svd --generate=header`

_CMSIS-SVD SVD Consistency Checker / Header File Generator V3.3.4
Copyright (C) 2010 - 2017 ARM Ltd and ARM Germany GmbH. All rights reserved.
Arguments: "ARM_Example.svd" --generate=header
*** ERROR M201: ARM_Example.svd (Line 63)
  Tag \<deviceNumInterrupts\> unknown or not allowed on this level.
*** INFO M317: ARM_Example.svd (Line 91)
  Interrupt \<description\> not set.
*** WARNING M208: ARM_Example.svd (Line 752)
  Ignoring \<dimIndex\> because specified \<name\> requires Array generation.
*** INFO M317: ARM_Example.svd (Line 771)
  Interrupt \<description\> not set.
*** INFO M317: ARM_Example.svd (Line 781)
  Interrupt \<description\> not set.
Found 1 Error(s) and 1 Warning(s)._

All versions of SVDConv (two Linux and one Win32) show the same behaviour.

Suggestions:

- Updating ARM_Example.svd to 1.3.3 schema standard, or commenting-out not yet supported constructs in current schema file.
- Providing versions of SVDConv in sync with lastest 1.3.3 schema version if the schema is to stay as is.

JonatanAntoni, 2017-03-28:
Thank you for your report. We will update SVDConv.

eperie, 2017-03-28:
Thank you for having taken this report into consideration.

JonatanAntoni, 2018-02-06:
Hi @eperie,

sorry for not reporting back here for long.
We have updated SVDConv to 3.3.13 in the meanwhile. This issue should be solved.
May I ask you to double check and close, please?

Thanks,
Jonatan

eperie, 2018-04-21:
Regressed in v5.3.0 - thanks.
Éric

closed 2018-04-21 by eperie