ARM-software/CMSIS_5 #172: Feature Request: enumeratedValueDataType pattern should allow binary literals

eperie, 2017-03-28:
A  **_lot_** of ARM silicon vendors datasheets do contain enumerated values for registers fields specified in binary.
Forcing users of CMSIS-SVD to convert such binary literals prior to fill enumerated values in a .svd files is going against the very purpose of schemes such as CMSID-SVD or IP-XACT I guess:

` <xs:simpleType name="enumeratedValueDataType">`
`    <xs:restriction base="xs:string">`
`      <xs:pattern value="[+]?(0x|0X|#)?[0-9a-fxA-FX]+"\>`
`    </xs:restriction>`
`</xs:simpleType\>`

Suggestions:
- augmenting modifying xs:pattern used for enumeratedValueDataType element to allow binary literals such as accepted in C++ standard or GCC/LLVM C extensions.
- enumerated values would then be rendered by SVDConv in hexadecimal with a comment containing their original binary values in order to enforce traceability and maintain compatibility with target C compilers that do not support the use of binary literals in C.

This could likely be achieved at a very low cost in a very near future, and would be extremely useful.

jkrech, 2017-03-28:
Are you aware that binary notation is already possible, however it uses a leading # instead of 0b?

eperie, 2017-03-28:
I was obviously not, thank you. Did I miss something in CMSIS-SVD.xsd, or in the .svd examples, or in some other documentation ?

closed 2017-03-28 by eperie

jkrech, 2017-03-28:
http://arm-software.github.io/CMSIS_5/SVD/html/svd_xml_conventions_gr.html

Constants

Number constants shall be entered in hexadecimal, decimal, or binary format.

    The Hexadecimal format is indicated by a leading 0x.
    The Binary format is indicated by a leading #.
    All other formats are interpreted as decimal numbers.
    The element <enumeratedValue>.<value> can be used to define constants.

eperie, 2017-03-28:
I see, my mistake then - I was expecting to see such an information in the schema file and did not look for it in the HTML documentation. Thank you.