ARM-software/CMSIS_5 #173: Questions about thread termination

Masterbar, 2017-03-29:
Hi,

I have few questions about thread termination:

1. Is it safe to call osThreadExit() from a detached thread?
2. What happens if I call osThreadTerminate() on a joinable thread before it exits? Will the thread still exists?
3. Does it make sense to call osThreadTerminate() with my own thread id at the end of the thread routine? Is it required?

Thank you,
Greg

RobertRostohar, 2017-04-04:
1. Yes, the thread will be terminated and its resources released.
2. Should behave similar as if the thread exits (osThreadExit or just returns). The thread will be terminated. The resources will be released either immediately (when join was active) or after another thread executes join or detach on that thread.
3. Not needed. Threads may exit with just returning.

Masterbar, 2017-04-19:
Thank you for the clarification.

closed 2017-04-19 by Masterbar