ARM-software/CMSIS_5 #174: RTOS : Questions about RTOSv1 license

ykabe, 2017-03-30:
Hi, I have a question about this repositry software license.
According to http://www.keil.com/pack/doc/CMSIS/RTOS/html/rtxImplementation.html, the software license of RTOSv1 is BSD.
However, according to LICENSE.txt, this repository software is Apache License 2.0.
the software license of "CMSIS_5 / CMSIS / RTOS / RTX / folder" is Apache License 2.0? Or BSD?

jkrech, 2017-03-30:
Sorry for the confusion. It is supposed to be Apache 2.0. We will update the source code header sections accordingly.

ykabe, 2017-03-30:
Thank you for your prompt response.
I understand that it is Apache 2.0.

JonatanAntoni, 2017-03-30:
Work is in progress. I give you feedback as soon as the update is done.

ykabe, 2017-04-04:
Thank you for your support.
When do you plan to complete the fix?

ykabe, 2017-04-04:
I'm sorry,I accidentally closed it.

JonatanAntoni, 2017-04-04:
We aim to resolve this issue with the upcoming CMSIS 5.1.0 release. Do you have time constraints here?

ykabe, 2017-04-04:
I would like to use RTX 4.82 on our products.
However, the code fix is near.
Is it possible to hurry to correct the RTOS folder?

JonatanAntoni, 2017-04-04:
Yes. I am going to address this soon. Its simply a matter of capacity at our side I have to manage.

ykabe, 2017-04-04:
Thank you for your support in your busy time.
We are expecting an early correction.

JonatanAntoni, 2017-04-05:
@ykabe I recently updated all the license comments for CMSIS-RTX on develop branch. May I ask you to review the changes? Please close this issue if your concerns are solved.

ykabe, 2017-04-05:
I confirmed that the change contents are satisfactory.
Thank you for your help despite how busy you are.

closed 2017-04-05 by ykabe