ARM-software/CMSIS_5 #176: arm_cfft_radix8_f32.c initialization

cezarymikowski, 2017-04-03:
There is 
arm_cfft_radix8_f32.c
procedure in
CMSIS_5/CMSIS/DSP/Source/TransformFunctions/
but for use it some internal structure must be initializeed: 
/*
* @brief  Core function for the floating-point CFFT butterfly process.
* @param[in, out] *pSrc            points to the in-place buffer of floating-point data type.
* @param[in]      fftLen           length of the FFT.
* @param[in]      *pCoef           points to the twiddle coefficient buffer.
* @param[in]      twidCoefModifier twiddle coefficient modifier that supports different size FFTs with the same twiddle factor table.
* @return none.
*/

How should it be done? I can not see arm_cfft_radix8_init_f32.c procedure (equivalent to arm_cfft_radix4_init_f32.c or arm_cfft_radix2_init_f32.c).

JonatanAntoni, 2017-04-03:
This file contains only the `arm_radix8_butterfly_f32` internal helper function. 
Why do you like to make use of this function directly? It is not intended to be an interface function from outside the DSP library.

cezarymikowski, 2017-04-04:
I missunderstood this function usage. Thank you for explanation. Closed.

closed 2017-04-04 by cezarymikowski