ARM-software/CMSIS_5 #177: Reimplementing RTX allocators using malloc/free - issue with mutex lock/unlock in SVC

bulislaw, 2017-04-03:
## Background
RTX allocators (https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_memory.c) require a static carveout of memory being allocated. That's causing troubles for platforms/systems that wish to use dynamic memory using heap -> malloc/free. In mbed OS we had to disable RTX dynamic memory to avoid dual carveout, basically having to have two separate heaps (which is a big issues on small targets).

## Solution
Using the fact that RTX allocators are defined as `WEAK` we decided to overwrite the functions, in mbed OS, with wrappers for malloc/free. https://github.com/ARMmbed/mbed-os/pull/4047

## Issue
Unfortunately this solution has one fundamental issue. Malloc and free calls are protected, by libc, with mutex, to make the them thread safe. For each compiler:
ARMC https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_lib.c#L594 
IAR https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/mbed_boot.c#L550
GCC https://github.com/ARMmbed/mbed-os/blob/feature_cmsis5/rtos/mbed_boot.c#L465

That doesn't work for RTX, as all the os*New calls, that will end up using malloc, are executed through SVC, at the same time, the mutexes are prevented to be locked in interrupt context: https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_mutex.c#L457

Can you please advice a way out of this situation?

JonatanAntoni, 2017-04-03:
This is a matter of how the libc callbacks for thread synchronization are implemented. Obviously for RTX we cannot make use of a RTX mutex because RTX itself is executed in handler mode (SVC interrupt).

What are the context malloc and free are allowed to be used?
- thread mode (for sure)
- RTX SVC interrupt (should be)
- arbitrary interrupts other then SVC (I assume not)

Thus it might be a valid solution to use osKernelLock and osKernelUnlock to prevent context switches instead of a mutex. These functions must not be called in interrupt context. Thus we have to skip lock/unlock if we are already in handler mode. To prevent misuse of malloc/free from other handlers then SVC, we might check for SVC handler as well.

May I ask you to try this alternative implementation and report back? Thank you.

ReinhardKeil, 2017-04-05:
We did further investigations and the standard C-library malloc/free alternative has the following implications compare to the local osRtxMemoryAlloc/osRtxMemoryFree implementation in [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_memory.c](rtx_memory.c):

- The standard malloc/free functions are also used by the user application and therefore protection against re-entrancy is required. This is typically achieved with a mutex, but some libraries i.e. [https://developer.arm.com/docs/dui0808/latest/the-arm-c-micro-library](ARM C Micro Library) may not offer the required functionality. This would need further investigation (i.e. GCC newlib-nano) as the limitation should be documented.

- The  C/C++ Standard Library Multithreading Interface implemented in [https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS2/RTX/Source/rtx_lib.c](rtx_lib.c) is protected against reentrancy with osMutex functions. But the usage of standard C library malloc/free functions inside the RTX functions itself implies that the protection with a standard RTOS mutex is no longer sufficient. Protection via mutex allows that other (high priority) threads can execute, but at the same time it needs to be ensured no other RTOS functionally is used that potentially allocates or releases memory. The  C/C++ Standard Library Multithreading Interface would need therefore be re-written to use critical sections based on osKernelLock/osKernelUnlock itself.

- As the C/C++ Standard Library Multithreading Interface maybe used also by functions like fopen/fclose there might be a significant time period where RTOS thread switching will be blocked when osKernelLock is used for protection. This may also happen when the malloc/free memory pool is fragmented. High priority threads can therefore no longer rely on deterministic execution which is a key element of a bare-metal RTOS implementation.

Compared to the implications listed above, the usage of the local osRtxMemoryAlloc/osRtxMemoryFree implementation only requires a dedicated memory area. The size is currently configured using the define OS_MEMPOOL_DATA_SIZE. This results in the declaration of the array os_mp_data that provides the memory pool. Currently the only drawback is this local memory pool that has a fixed size which implies that the size configuration is a burden to the user.  Does osRtxMemoryAlloc/osRtxMemoryFree have any other drawbacks?

bulislaw, 2017-04-05:
Thanks for detailed analysis. I agree that, for applications that do one single (predictable) thing, and the amount of dynamic memory can be configured upfront, having a RTX carveout is not an issue. The RTX memory can be tweaked in such a way that it suits the app and memory is not wasted, even on very constrained targets. Unfortunately for more generic usecases, like mbed OS, having, up  to, couple of K reserved at compile time for RTX usage is a problem. Not only we can't predict how much memory could be needed, but also we can't really rely on our users to go into nitty gritty and set the memory to required levels. A lot of our users just don't know/care about things like RTX memory models and how it conflicts with standard (heap) dynamic memory.
Right now we have the RTX dynamic memory disabled and we require users to provide their own backing storage for RTX objects, of course it's not and ideal solution, but in my opinion it may be better than having two heaps.

ReinhardKeil, 2018-05-24:
closing as no further activity

closed 2018-05-24 by ReinhardKeil