ARM-software/CMSIS_5 #178: missing a generic CMSIS version macro

houssema, 2017-04-03:
Hi,

The CMSIS version is defined in every core file by the following "__CMX_CMSIS_VERSION", is there any possibility to add CMSIS version macro independent from the used core ? 

This will make easy to get the CMSIS version (V4 or V5) for software implemented to run on all supported cores. 

With the actual implementation we have to do the following:

/* include core specific file */
#if defined (CORTEX_M7)
#include "core_cm7.h"
#if defined (CORTEX_M4)
#include "core_cm4.h"
#elif defined (CORTEX_M3)
#include "core_cm3.h"
#elif defined (CORTEX_M0)
#include "core_cm0.h"
#elif defined (CORTEX_M0PLUS)
#include "core_cm0plus.h"
#endif

/* check the CMSIS version */
#if(__CM7_CMSIS_VERSION_MAIN == 5) || (__CM4_CMSIS_VERSION_MAIN == 5) || (__CM3_CMSIS_VERSION_MAIN == 5) || (__CM0_CMSIS_VERSION_MAIN == 5)  || (__CM0PLUS_CMSIS_VERSION_MAIN == 5) 

/* We have CMSIS V5 */
...
#else 

/* We have CMSIS V4 */
...
#endif 

Thanks and regards

JonatanAntoni, 2017-04-04:
Thank you for this proposal. I am going to address this soon.

JonatanAntoni, 2017-04-05:
Hi @houssema. The solution I would suggest is adding a new cmsis_version.h
```
#define __CMSIS_VERSION_MAIN  ( 5U)                                  /*!< [31:16] CMSIS HAL main version */
#define __CMSIS_VERSION_SUB   ( 0U)                                  /*!< [15:0]  CMSIS HAL sub version */
#define __CMSIS_VERSION       ((__CM0_CMSIS_VERSION_MAIN << 16U) | \
                                    __CM0_CMSIS_VERSION_SUB           )  /*!< CMSIS HAL version number */
```
and update all the core_*.h files accordingly to
```
#define __CMx_CMSIS_VERSION_MAIN  __CMSIS_VERSION_MAIN
#define __CMx_CMSIS_VERSION_SUB   __CMSIS_VERSION_SUB
#define __CMx_CMSIS_VERSION       __CMSIS_VERSION
```
to be backward compatible. But I would discourage the __CMx macros.
Does this solve your concerns?

houssema, 2017-04-18:
Hi @JonatanAntoni. 

Thank you for your proposal, I agree with this implementation, it will help with the next CMSIS versions.

I have just a comment about __CMSIS_VERSION definition, it should be as below:
`#define __CMSIS_VERSION       ((__CMSIS_VERSION_MAIN << 16U) | \
                                __CMSIS_VERSION_SUB)             /*!< CMSIS HAL version number */`

JonatanAntoni, 2017-04-18:
Hi @houssema, 

Thanks for your feedback. I corrected this definition in my comment above, was just a typo.

JonatanAntoni, 2017-04-19:
As we are currently working on Core(A) we should rather go for
```
#define __CM_CMSIS_VERSION_MAIN  ( 5U)                                  /*!< [31:16] CMSIS HAL main version */
#define __CM_CMSIS_VERSION_SUB   ( 0U)                                  /*!< [15:0]  CMSIS HAL sub version */
#define __CM_CMSIS_VERSION       ((__CM_CMSIS_VERSION_MAIN << 16U) | \
                                   __CM_CMSIS_VERSION_SUB)                 /*!< CMSIS HAL version number */
```
The Core(A) version will be `__CA_CMSIS_VERSION` respectively. Okay @houssema?

JonatanAntoni, 2017-04-27:
Closing this issue due to no further discussion happened.

closed 2017-04-27 by JonatanAntoni

houssema, 2017-04-27:
Thanks @JonatanAntoni for this implementation, and sorry for the late reply