ARM-software/CMSIS_5 #179: Missing seperate CAN Bus Off define

aph-le, 2017-04-05:
In the Driver_CAN.h file there is a define:
ARM_CAN_UNIT_STATE_INACTIVE (0U) ///< Unit state: Not active on bus (initialize or error bus off)
In most cases "bus off" and "initialize" are two diffrent states in the CAN controller, which might come together, but don't have to. Initialize is not an error in the CAN controller but bus off is. To distinguish between the to state there should be 2 diffrent defines for that. Like:
ARM_CAN_UNIT_STATE_INITIALZE
ARM_CAN_UNIT_STATE_BUS_OFF

JonatanAntoni, 2017-09-01:
Hi @aph-le,

Excuse me to have taken such long time, please.
We discussed the issue intensively and now decided to enhance the CAN Driver API according to your suggestions, soon.

Cheers,
Jonatan

JonatanAntoni, 2017-09-19:
Hi @aph-le,

may I ask you to check the changes we recently contributed to the CAN API?
Does this already solve your concerns? Or is anything still to be discussed?

Cheers,
Jonatan

JonatanAntoni, 2017-10-17:
Closing this issue because it should be solved. Please feel free to reopen it if you have further concerns.

closed 2017-10-17 by JonatanAntoni