ARM-software/CMSIS_5 #181: EXTI in the CMSIS Driver

809205580, 2017-04-10:
Excuse me, Why not CMSIS Driver support EXTI?

JonatanAntoni, 2017-04-10:
Hi @809205580. We have currently no CMSIS Driver API for GPIO related operations defined. May I ask you to provide us some use cases you would like to have such an API for? What should the API look like to be useful?

809205580, 2017-04-10:
I have used GPIO driver(used by RTE Drivers for STM32F1), but I didn't find the operation associated with the interrupt used for button or key.

JonatanAntoni, 2017-04-10:
Please be more specific. Which GPIO driver do you use? I assume the STM32Cube one? This is the native driver API provided by STM, not a CMSIS one.

The STM32Cube (and the older STM32Peripheral Library) provide some support for EXTI configuration. If you have detailed questions about the STM32 library implementation please consider to contact the ST community. I am glad to discuss CMSIS related issues here.

JonatanAntoni, 2017-04-19:
@809205580 do you have any further concerns about EXTI handling with CMSIS? Otherwise I would kindly ask you to close this issue please.

809205580, 2017-04-19:
OK.

JonatanAntoni, 2017-07-17:
Closing due to no further open questions to be discussed. Please feel free to reopen if you still have any concerns.

closed 2017-07-17 by JonatanAntoni