ARM-software/CMSIS_5 #182: CMSIS_os.h (specification) differs from the implementation

Milacus, 2017-04-12:
Hey there,

I was just wondering if I got something wrong or if there might be a bug in the CMSIS_os.h (which serves as specification, right?) of the RTOS.

If you have a look at the [CMSIS_os.h, line #138](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS/RTX/INC/cmsis_os.h#L138), it specifies that the thread function will have a `void const *argument`. 
But if you look at [rt_CMSIS.c, line #777](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/RTOS/RTX/SRC/rt_CMSIS.c#L777) it expects a `void *argument` which will later on be passed to the thread's function. Every further step of the implementation will work with the not constant pointer.

If it's intended to fix this there might be some more work and deeper knowledge necessary as I'm able to offer.

Greets,
Milacus

JonatanAntoni, 2017-04-12:
Hi @Milacus, thanks for your feedback. 
RTX is no longer under active development. It has been superseded by RTX5 a while ago.

For RTX4 you are right. We take a non-const pointer for the argument on thread creation but make it const when handing it over to the actual thread function. In this data flow direction adding const is totally legal. No one except the thread function itself uses this argument. 
For RTX5 we decided to change the thread function prototype to take a non-const pointer as argument. 

Does this slight inconsistency in RTX4 results in any serious issue for you?

Milacus, 2017-04-12:
Thanks for the quick and helpful replay @JonatanAntoni . 

It doesn't cause any errors. I just came up with some warnings in one of my projects and found this "issue" but it's no problem as is did already adapt my software accordingly.

JonatanAntoni, 2017-04-12:
It should not give you warnings unless you use the pointer inside the thread function to modify the pointed argument.. Can you please let me know about the warnings you get in detail?

Milacus, 2017-04-12:
***warning: passing 'const void*' to parameter of type 'void *' discards qualifiers***

where I'm calling `osThreadCreate`

But like I said. With the small adjustment of dropping the `const` I was able to solve this

JonatanAntoni, 2017-04-13:
Ah yes, sure. Due to the interface of osThreadCreate does not respect this qualifier, you cannot use a const argument here. If this is not a big issue for you I would stick to the current implementation. As I mentioned this RTX4 implementation has been superseded by RTX5 and the problem is solved there.

Would it be a solution for you to upgrade to RTX5? There is a CMSIS-RTOS v1 compatibility layer available to limit the migration work on the short run. Take a look into our [RTX Migration Guide](https://www.keil.com/pack/doc/CMSIS/RTOS2/html/os2MigrationGuide.html).

Milacus, 2017-04-13:
We were working on the migration but ran into some problems.

Right now we gave other issues a higher priority and put the RTOS migration on hold. As soon as some time opens up I'll have another look into it.

Thanks a lot for your help and the tip

closed 2017-04-13 by Milacus