ARM-software/CMSIS_5 #183: osKernelLock() and osDelay()

juliobc, 2017-04-18:
Hi,

I want to make a delay inside a OS critical section in RTOS2. When I lock the kernel, the tick count doesn't stop and I try an osDelay() but it fails. I know that the task switches are lock and then the Idle Thread doesn't go to running state while my thread is waiting, but the return of osDelay() is osOK. It is correct? 

I try the next solution:

`
uint32_t tick_time = osKernelGetTickCount();
while( osKernelGetTickCount() < (tick_time + 500) ){}
`

I think that it is a good workaround if I supose that the tick doesn't overflow

JonatanAntoni, 2017-04-18:
Hi @juliobc,

Thank you for your feedback. I am going to clarify whether osDelay and osDelayUntil should return a error code when used with kernel locked. 

osDelay cannot be used inside a OS critical region, i.e. when the kernel has been locked. The semantics of osDelay is to put the running thread to a wait list and resume execution with another thread. Inside the OS critical region you have to use busy waiting as you stated.

Please be aware that the tick will overflow regularly, depending on the time-per-tick you use. You might use `while( osKernelGetTickCount() != (tick_time + 500) ) { }` instead.

juliobc, 2017-04-18:
Hi @JonatanAntoni , Thank you for your reply.

Inside a OS critical region, it's posible to execute a ISR code?? If a code of ISR takes more than 1 mseg, Could I lost the condition osKernelGetTickCount() != (tick_time + 500) ??

JonatanAntoni, 2017-04-18:
This is implementation dependent. For RTX a OS Critical Section (i.e. osKernelLock) only prevents the scheduler from doing context switches. All ISR operations are left untouched. Thus you might lose the tick, yes.
Can you provide us some more details of what your are trying to archive? Why do you need to keep the kernel locked while waiting?

juliobc, 2017-04-18:
ok,
I just studying funtionality. My goal is get the very low power with a STM32F4 and RTOS2. I am studying how to make a Stop Mode in the Idle task without break the funtionality of independents and differents threads.

JonatanAntoni, 2017-04-18:
Ah, okay. You'd like to read our [Low-Power Operations](http://www.keil.com/pack/doc/CMSIS/RTOS2/html/theory_of_operation.html#TickLess) section. To suspend the kernel during tick-less sleep you can use [osKernelSuspend](http://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__KernelCtrl.html).

JonatanAntoni, 2017-04-25:
@juliobc did my references to the documentation help you out? May I ask you to close this issue if you do not have further concerns, please?

juliobc, 2017-04-25:
Hi @JonatanAntoni 

I have another question related with the low power operation. It is valid the next implementation for the osRtxIdleThread?:
`  
  #define MINIMUM_TIME_TO_DEEP_SLEEP 2 /* 2 msec */
  for (;;) {
    tc_wakeup = osKernelSuspend();
    /* Is there some time >= MINIMUM_TIME_TO_DEEP_SLEEP to sleep? */
    if (tc_wakeup >= MINIMUM_TIME_TO_DEEP_SLEEP) {
      tc = 0;
      /* Enter the low power state */
      MSP432_LP_Entry();
      __WFI();
    /* Adjust the kernel ticks with the amount of ticks slept */
    osKernelResume (tc);
    }
    else {
      osKernelResume (0);
      __WFI();
    }
  }`

Is there any problem with osKernelResume(0)? Could any interrupt jump while I check the condition (tc_wakeup >= MINIMUM_TIME_TO_DEEP_SLEEP) and then __WFI() lost the interrupt?

JonatanAntoni, 2017-04-25:
Why do you call `__WFI()` after `osKernelResume(0)`?
`osKernelResume(0)` should be no problem if you resume without a delay. Or lets say with a minimum delay << 1 tick.
I admit that I cannot guarantee that this simple example does not loose any interrupts. Let me investigate here. I will report back later.

juliobc, 2017-04-25:
Actually I use __WFE() instead of __WFI().

when the delay is very short I want the osRtxIdleThread implementation looks like this:
`void osRtxIdleThread (void *argument) {
  (void)argument;
  for (;;) {
     __WFE();
  }
}`

JonatanAntoni, 2017-04-25:
Ah, okay. So you simply go to sleep but don't disable the systick in this case. Looks valid.

juliobc, 2017-04-27:
it don't run correctly.

`tc_wakeup = osKernelSuspend();
if (tc_wakeup > 5) {
...;
osKernelResume(tc);
}
else
{
osKernelResume(0);
__WFE();
}`

Could it be that osKernelResume(0); causes __WFE () to exit instantly?

a delay of 5 msec is running about 14 msec

JonatanAntoni, 2017-04-27:
Yes, for sure. The ordering must be

`tc_wakeup = osKernelSuspend();
if (tc_wakeup > 5) {
// setup low power timer / rtc to generate an interrupt/event after tc_wakeup milliseconds
__WFE();
osKernelResume(tc);
}
else
{
// don't go to sleep at all
osKernelResume(0);
}`

`__WFE()` returns after the next event. If you call osKernelResume prior to `__WFE()` the next SysTick_Timer interrupt wakes up the device again.

Please note that I am currently working on an update to the example.

juliobc, 2017-04-27:
If the next SysTick_Timer interrupt wakes up the device again then I think that there is no problem. The problem is when `__WFE()` returns with another kind of exception like a pending SVCall for example.

JonatanAntoni, 2017-04-27:
An SVCall should not happen, because this is only used for calls issued by your application. Other interrupts, beside PendSV, are not used by RTX. PendSV in turn is used when you use RTOS functions from your own interrupt routines, i.e. unblocking a waiting thread. The conditions __WFE wakes up are device specific and typically configurable. So you have to assure which interrupts should wake up the device from suspend.

juliobc, 2017-04-27:
I notice that te next implementation inside the Idle Thread makes the delays 3 times slower:
`for(;;){ tc_wakeup = osKernelSuspend(); osKernelResume(0); }`
Is it possible?

JonatanAntoni, 2017-04-27:
Yes, this might happen because each call to osKernelSuspend causes a minimal jitter (<< 1 tick) to the system tick. This is due to switching off the timer clock for a few cycles.

Why do you want to suspend the kernel for 0 ticks at all? The idea behind the given tc_wakeup is that you really delay execution until the next thread becomes ready again. The jitter should not matter that much as long as the kernel suspension time is much larger.

juliobc, 2017-04-28:
I want to go into low power only if the tc_wakeup is greater than a minimum time_msec, otherwise the drivers may fail (turn off the DMA).
A great solution is if I had a function to get the tc_wakeup without suspending the kernel, and then decide whether to suspend the kernel or not.

juliobc, 2017-04-28:
I found a workaround: make a delay without low power:

`for(;;){ tc_wakeup = osKernelSuspend(); if (tc_wakeup > 5) {low_power_delay(tc_wakeup);}else {delay_nops(tc_wakeup);}osKernelResume(tc_wakeup); }`

JonatanAntoni, 2017-04-28:
Hi @juliobc. Okay, so in fact you might have an ongoing DMA transfer active? 

The available sleep modes are device specific. So it is device specific which sleep modes leave the peripherals and memories untouched. Unfortunately I am not able to provide a low power example for all the possible use cases you might face with.

Use busy waiting might help you out here, yes. Using `tc_wakeup` as an input to `osKernelResume()` might be to much simplified. You should rather provide the actual time the device was suspended (either by using `__WFE()`, `__WFI()` or `delay_nops`) to `osKernelResume()`. This is why we suggest the usage of an explicit low power timer or real time clock to keep track of time while the system tick is suspended.

I hope this helps you out for your actual project.

juliobc, 2017-04-28:
Hi @JonatanAntoni 
yes. Thank you for your quick support.

JonatanAntoni, 2017-05-02:
You're welcome. Please close the issue if you do not have further aspects to be discussed.

closed 2017-05-02 by juliobc