ARM-software/CMSIS_5 #184: Issue with osMutexPrioInherit

Masterbar, 2017-04-19:
Hi,

The OS is stuck in an infinite loop in svcRtxMutexRelease() when I do the following sequence:

`
const osMutexAttr_t Thread_Mutex_attr = {
  "myThreadMutex",                          // human readable mutex name
  osMutexRecursive | osMutexPrioInherit,    // attr_bits
  NULL,                                     // memory for control block   
  NULL                                      // size for control block
  };
  
osMutexId_t CreateMutex (void)  {
 
  mutex_id = osMutexNew(&Thread_Mutex_attr);
  if (mutex_id != NULL)  {
    // Mutex object created
  }   
  return mutex_id;
}

...

  osMutexId_t m0, m1;
  m0 = CreateMutex();
  m1 = CreateMutex();
  osMutexAcquire(m0, 0);
  osMutexAcquire(m1, 0);
  osMutexRelease(m0);
  osMutexRelease(m1);
  osMutexAcquire(m0, 0);
  osMutexAcquire(m1, 0);
  osMutexRelease(m0);  // We are stuck here
  osMutexRelease(m1);
`
We are then in this infinite loop in svcRtxMutexRelease():
`    while (mutex0) {
      // Mutexes owned by running Thread
      if ((mutex0->thread_list != NULL) && (mutex0->thread_list->priority > priority)) {
        // Higher priority Thread is waiting for Mutex
        priority = mutex0->thread_list->priority;
      }
      mutex0 = mutex0->owner_next;
    }
`

Without the attribut osMutexPrioInherit everything is fine.

JonatanAntoni, 2017-04-19:
Hi @Masterbar, thank you for your report. We will investigate this issue.

JonatanAntoni, 2017-04-19:
I fixed this issue in [729f6ab](https://github.com/ARM-software/CMSIS_5/commit/729f6ab08540342b7de7612fac103539d5b2f168).
@Masterbar can you please verify that the provided fix works for you? Please close this issue on success.

Masterbar, 2017-04-19:
It's nice to see such quick feedback and fix :-)
I tested it and it works, thank you!

closed 2017-04-19 by Masterbar

eugene345, 2017-08-25:
Hello @JonatanAntoni ,

I'm on RTX 5.2.0 and I can see that the fix is there. But my program is stuck in the same loop in svcRtxMutexRelease(). The program is quite complicated, so, I'm not sure, how to isolate and provide some code for you to simulate the bug. The code snippet, provided by @Masterbar, works without errors.

Best regards,
Eugene.

JonatanAntoni, 2017-08-30:
Hi @eugene345,

It's hard for me to analyse the issue without having a reproducer to debug at hand.

May I ask you to dig a little deeper, please? What is the runnig_thread->mutex_list is about? 

There must be a cyclic link. If you can figure out which other mutexes are currently owned you might get an idea which one causes the cyclic link. Next we have to find out whether the cyclic link is caused by an bad chain or unchain operation.

Perhaps we can construct a reproducer if we can retrace the mutex operations. Do you have the possibility to activate the EventRecorder and record all the mutex operations? Such an event trace might be very helpful.

Best,
Jonatan

eugene345, 2017-08-30:
Hi @JonatanAntoni ,

I tried to find the problem by myself. But I haven't clear knowledges about CMSIS RTOS internal structure and implementation.  And I can't spend too much time on this bug at this moment.
So, I've attached some files (Event recorder log file and screenshot of local variables). I hope, it could help.
I'm here, if you need something extra.

Best regards,
Eugene.

[mutex_bug.zip](https://github.com/ARM-software/CMSIS_5/files/1263618/mutex_bug.zip)

JonatanAntoni, 2017-08-30:
Hi @eugene345,

The trace looks helpful.
I guess the trace ends because releasing mutex C14 causes the endless loop?

According to your screenshot the running thread still holds at least mutexes AFC and A8C. To see more you have to expand owner_next field recursively until this field is 0. If you follow this mutex chain you should see a cyclic chaining which is baaaad. Which mutexes do you expect the running thread is holding at that point in time?

The trace tells me in event 777 that mutex A8C should have been released. So is must not be member of this list any more. Looks wrong. Are you able to put a breakpoint to the mutex release that causes event 777? If you step into the RTX code you should see this mutex to be removed from the running threads mutex_list. Perhaps somethings going wrong here.

Are you able to step through your code starting at event 804? Which mutexes are held by the thread at this point in time? Does the mutex_list reflect your expectation?

Unfortunately I cannot tell you much more without being able to debug on my own. It seems you are using MDK/uVision? So if you mind to provide some reproducing code here in public you might consider to get in touch with MDK support.

Cheers,
Jonatan

eugene345, 2017-08-30:
Yes, mutex C14 causes the endless loop.
I've tried to expand owner_next field, but it doesn't end with 0, it expands again and again.
Not sure, which mutexes the running thread is holding. Probably, A8C.
The problem that this bug occurs sporadically. And during execution of a complicated function (file downloading from http server and saving to eMMC memory). There is no way to set a breakpoint, because I don't know in advance at which iteration it will stuck. Another problem, that this code runs on a custom board, so, no sense to provide any code.
I think, for now I'll try to find a workaround, because I can't delay the project anymore.
And later, I'll play with mutexes and try to simulate the bug to provide a reproducer for you.

Best regards,
Eugene.

JonatanAntoni, 2017-08-30:
Hi @eugene345,

Yes, I already expected that it expands again and again. And that's the problem. You have to look where the cyclic chaining happens. This chain is a simple linked list. There must not be any element in the chain that points to an element which is already part of the list before, thus forming a cycle. We have to figure out which element forms the cycle, i.e. has a bad owner_next pointer. 

Does your custom board have any instruction trace capabilities? This might help as well to retrace the execution flow before you stuck. Another possibility might be using data breakpoints if you can nail down the malicious pointer value in order to find the code place it is written by.

I feel sorry that I can not do much more to help at the moment.

Cheers,
Jonatan