ARM-software/CMSIS_5 #185: call to osKernelGetState() with osKernelState=osKernelReady

juliobc, 2017-04-19:
Hi,

Let me tell you my case: I use STM32F4 and its HAL libraries with RTOS2. HAL libraries aren't ready to work with RTOS completely. One problem is that they use the systick timer for count ticks, but when you add RTOS the systick timer is not free and then the HAL libraries lost its tick count.

I thought the next implementation for workaround the issue:

`uint32_t HAL_GetTick(void){
	return osKernelGetTickCount();
}
`

The problem is that HAL libreries use the HAL_GetTick() before that Kernell is running and then I have a Hard fault interrupt.

Another workaround is:

`uint32_t HAL_GetTick(void)
{
	osKernelState_t st = osKernelGetState();
	if ( st > osKernelReady )
		return osKernelGetTickCount();
	else
		return 0;
}`

and now there are no problem. My question is, It is a good workaround? Another workaround?

Thank you

RobertRostohar, 2017-04-19:
Take a look at:
http://www.keil.com/forum/62308/

juliobc, 2017-04-19:
ohhh, thank you, I had not found a solution. Great solution!

closed 2017-04-19 by juliobc