ARM-software/CMSIS_5 #188: CMSIS DSP: arm_math warnings for ARMCC 5.2x

0xc0170, 2017-04-25:
This is the output if ARMCC is used with ``--gnu`` flag (thus ``__GNUC__`` is defined). This is causing the following warnings:

```
[Warning] arm_math.h@298,0:  #2803-D: unrecognized GCC pragma
[Warning] arm_math.h@299,0:  #2803-D: unrecognized GCC pragma
[Warning] arm_math.h@300,0:  #2803-D: unrecognized GCC pragma
[Warning] arm_math.h@301,0:  #2803-D: unrecognized GCC pragma
[Warning] arm_math.h@4949,0:  #3108-D: attribute "__packed__" does not apply here
[Warning] arm_math.h@4957,0:  #3108-D: attribute "__packed__" does not apply here
[Warning] arm_math.h@7218,0:  #2803-D: unrecognized GCC pragma
```

The pragma are easy to fix - add "not defined ARMCC" : ``#if defined(__GNUC__) && !defined ( __CC_ARM )``.  I'll can send a patch for review

However, how  to fix the ``__packed__`` attribute?

I have installed armcc 5.23 that I am testing with.

0xc0170, 2017-04-25:
Proposed fix : https://github.com/ARM-software/CMSIS_5/pull/189

christophe0606, 2019-05-29:
There were several changes to arm_math.h since this issue was opened.

I'll think this issue is fixed. Feel free to reopen it if you still have the problem.

closed 2019-05-29 by christophe0606