ARM-software/CMSIS_5 #193: Requesting to have interfaces defined for ARM MPU.

cashq, 2017-04-28:
Currently in core_cm7.h there’re only access macros, no any API provided. It would be helpful, if we can have some functions implemented, like for region configuration:

```
void MPU_configure_region(uint32_t region, uint32_t base, uint32_t rasr, bool enable)
{
      __disable_irq();
      SCB_DisableDCache();
      SCB_DisableICache();
 
      MPU->RNR = MPU_RNR_REGION_Msk & region;               // Region to configure
      MPU->RBAR = base & MPU_RBAR_ADDR_Msk;                 // Set base address for region
      MPU->RASR = rasr;                                     // Set region size and settings
      if(enable)
            MPU->CTRL = MPU_CTRL_PRIVDEFENA_Msk | MPU_CTRL_ENABLE_Msk;  // Enable MPU
 
      SCB_EnableICache();
      SCB_EnableDCache();
      __DSB();
      __ISB();
      __enable_irq();
}
```

JonatanAntoni, 2017-04-28:
Hi @cashq. We are currently working on this as part of CMSIS-Zone. Please be patient until we have a first proposal available to be discussed here.

Regarding your proposal we have already discussed having such elaborated functions. In terms of CMSIS-Zone we expect such a method to be called multiple times on each thread context switch. This is because we aim to use the MPU to protect threads (stacks and private data for instance) against each other. Thus we need to reconfigure the MPU on each context switch.

We consider the execution time to be crucial, i.e. measured as instructions needed. Thus we are looking for a more compact solution. Currently we are thinking about providing the exact register values needed for each region as an MPU Table, i.e. array of register values. The reconfiguration would be a matter of a "mem copy" only, i.e. dumping the table directly into the MPU registers. We expect that this would save a major amount of instructions needed.

What do you think? We would be glad to see more real world cases MPU protection is used in.

cashq, 2017-05-02:
Thank you, @JonatanAntoni. It would be better to have the compact solution. However, I hope we could have a 'basic' solution soon, so that we can use the officially provided interfaces to enable/disable/configure ARM MPU.

OussemaHajjem, 2017-05-10:
Hi @JonatanAntoni ,  Please find below a proposal for the MPU API implementation:

    #if defined (__MPU_PRESENT) && (__MPU_PRESENT == 1U)
	
    /**
        \brief  Enables the MPU
        \param [in]  control: Specifies the control mode of the MPU during hard fault, 
                NMI, FAULTMASK and privileged access to the default memory 
                The parameter control is assigned to the field MPU->CTRL[2:1]
      */
    __STATIC_INLINE void MPU_Enable(uint32_t control)
    {
      __DSB();
      __ISB();
    
      MPU->CTRL   = control | MPU_CTRL_ENABLE_Msk;
      SCB->SHCSR |= SCB_SHCSR_MEMFAULTENA_Msk;
    
      __DSB();
      __ISB();
    }
    
    /**
        \brief  Disables the MPU
      */
    __STATIC_INLINE void MPU_Disable(void)
    {
    
      __DSB();
      __ISB();
    
      SCB->SHCSR &= ~SCB_SHCSR_MEMFAULTENA_Msk;
      MPU->CTRL  &= ~MPU_CTRL_ENABLE_Msk;
    
      __DSB();
      __ISB();
    
    }
    
    /**
      \brief  Configures a memory region to be protected.
      \param  region [in]: MPU region number to configure
      \param  base [in]: base address of the MPU region
      \param  rasr [in]: MPU attributes of the MPU region
      */
    __STATIC_INLINE void MPU_ConfigRegion(uint32_t region, uint32_t base, uint32_t rasr)
    {
      MPU->RNR = region;
      MPU->RBAR = base;
      MPU->RASR = rasr ;
    }
    #endif /* defined (__MPU_PRESENT) && (__MPU_PRESENT == 1U) */

I hope this will be helpful

JonatanAntoni, 2017-05-11:
Thank you @OussemaHajjem for your contribution.

The ARM MPU for ARMv7 devices supports the configuration of up to 4 regions by consecutive memory writes. I can imagine that exploiting this feature may improve the execution time needed. As we aim to reconfigure the MPU on each RTOS thread switch execution time needed for MPU reconfiguration seems to be crucial for me. What do you think?

OussemaHajjem, 2017-05-16:
Hi @JonatanAntoni, I agree with you 
In addition, it will be helpful if you provide an API that configures a specific MPU region, this will give more flexibility for applications running without RTOS.

cashq, 2017-06-19:
Hi guys, may I know when it will be available in the ARM header file?

JonatanAntoni, 2017-06-19:
Hi @cashq,

I am working on the integration of the proposals. It will take me some additional time due to the vacation period. Do you have a deadline we should consider?

Kind regards,
Jonatan

cashq, 2017-06-19:
Hi @JonatanAntoni, I don't have a deadline for that, at least for now. Just want to know the schedule. : )

JonatanAntoni, 2017-07-19:
Hi all,

I added a first proposal for ARMv7 MPU implementation with f71f6cf.
Your feedback is highly appreciated.

Cheers,
Jonatan

JonatanAntoni, 2017-08-29:
Hi all, 

ARMv7 MPU implementation has been released with CMSIS 5.1.
ARMv8 MPU implementation is available on the development branch.

Do you have any feedback regarding the implementation? Otherwise I would like to close this issue as it has been resolved, @cashq.

Cheers,
Jonatan

JonatanAntoni, 2017-10-17:
I close this issue now. Please feel free to reopen if further issues arise.

closed 2017-10-17 by JonatanAntoni