ARM-software/CMSIS_5 #196: SVDConv.exe: __MPU_PRESENT macro is not in generated header file

David-Garcia-Polo, 2017-05-02:
For a cortex M0, the SVDConv.exe v3.3.9 generated header does not contain the macro __MPU_PRESENT defined to 0 , as it did in older versions (when the macro disappeared is not known).

Even though no CM0 has MPU, when writing portable code for different devices will healp greatly to only have to check one macro and not several.

Also an issue with compatibility of old code.

JonatanAntoni, 2017-05-02:
Hi @David-Garcia-Polo, you might check the availability of MPU using `#if defined(__MPU_PRESENT) && (__MPU_PRESENT == 1)`. Does this help?

David-Garcia-Polo, 2017-05-02:
It heelpas with Issue 1 (having to check different macros) but not with the old code compatibility.

simtind, 2017-12-15:
Has there been any progress on this issue?

JonatanAntoni, 2017-12-15:
Hi @simtind,

unfortunately the change has not been implemented due to resource limitation, yet.

The currently discussed solution is to make those feature tags optional. If they are not specified, nothing will be generated. But if you specify for instance `<mpuPresent>0</mpuPresent>` the according `#define __MPU_PRESENT 0` shall be generated.
That would give you better control over the generation result.

Please give us some additional time to update SVDConv.

Best,
Jonatan

simtind, 2017-12-15:
Sounds like a goood solution, thank you.