ARM-software/CMSIS_5 #197: SVDconv.exe: derivedFrom peripherals do not use same headerStructName

David-Garcia-Polo, 2017-05-02:
headerStructName tag is used to define the name of the struct to be use in the peripheral struct. Derived peripherals using the derivedFrom attribute do not inherit the headerStructName of the "master" peripheral, and cannot use headerStructName tag themselves with the same name since error M372 is thrown.

*** ERROR M372: nrf51.svd (Line 8846)
  Peripheral <headerStructName> 'TIMER' already defined (Line 8300)

Present in SVDconv.exe version 3.3.9

thorstendb-ARM, 2017-05-03:
Hi David,

can you please give an example .svd file?
Please also compare if the example I attached should trigger your error.

David-Garcia-Polo, 2017-05-03:
Your file produces the error. As it can be seen in the header generated two structs are generated for the peripherals, ClusterHeaderStructNameTest_Type and ClusterHeaderStructNameTest2_Type. Only one should be generated, ClusterHeaderStructNameTest_Type, and both peripherals should use it.

David-Garcia-Polo, 2017-05-03:
[nrf51.zip](https://github.com/ARM-software/CMSIS_5/files/972706/nrf51.zip)

Example of file as requested. Check peripherlas TIMER1 and TIMER2, or RTC1 or others.

simtind, 2017-12-06:
Has there been any progress on this issue?

JonatanAntoni, 2017-12-06:
Hi @simtind,

unfortunately there is no progress, yet.
Thanks for reactivating this topic.

Best,
Jonatan

jkrech, 2017-12-06:
Unfortunately we have not had time to resolve the issue.
One thing I noticed in your description is the fact that you are re-specifying 

      <size>32</size>
      <access>read-write</access>

in the derived peripherals TIMER1 and TIMER2. Doing so results in individual types being generated instead of sharing the same peripheral type.
Once you remove these tags and you remove/comment 

`<headerDefinitionsPrefix>NRF_</headerDefinitionsPrefix>`

 you will find that TIMER0, TIMER1 and TIMER2 share a single type: Timer_Type

So it seems that the 

`<headerDefinitionsPrefix>`

 is causing the trouble, as it results in the types:
NFR_Timer_Type for TIMER0 but unfortunately for TIMER1 and TIMER2 the NRF prefix is added a second time: NRF_NRF_Timer_Type.

We are currently reviewing when we can schedule a fix and will provide an update here.

simtind, 2017-12-06:
Ok, good to have an update. We'll look at preventing the repeated size and access tags on our end.

JonatanAntoni, 2018-02-06:
Hi @David-Garcia-Polo,

The issue should have been solved with update to SVDConv 3.3.13.
May I ask you to double check and close the issue, please?

Thanks,
Jonatan

simtind, 2018-02-12:
Hi, David is out on leave over the summer, but I can confirm it's working on my end after the required modifications as per @jkrech.

JonatanAntoni, 2018-02-12:
Thanks for confirmation, @simtind.

closed 2018-02-12 by JonatanAntoni