ARM-software/CMSIS_5 #198: New feature: SVD new tag alternateEnumeratedValue

David-Garcia-Polo, 2017-05-02:
Is it possible to get a new tag alternateEnumerated value to define a different name enumerated value with the same value as another one, and SVDConv not throwing Warning M333?

JonatanAntoni, 2017-05-02:
Relates to #15.

sgrange, 2018-03-08:
Jonatan,

Thanks for your response.  I would like to add to the discussion our thoughts on the value of using multiple enumeration entries with the same enumeration value.

We have peripherals that have different modes of behavior (e.g. master and slave modes).  Some register fields have different meaning when you are in one mode vs. the other.  We would like to have enumerations to clearly select options within the field that are logical for the mode you are in.  Thus the need for multiple names that point to the same enumerated value.  One for each mode.  

As mentioned before, temporary support for deprecated names would also be a use model.

We tried several other ways around the issue as such:
1) Tried to have two differently named fields with the same bit range.  This would produce two enumeration types which is acceptable and probably preferred.  However this produced errors in the conversion.
2) Tried to have two differently named registers with the same address offset.  This also produced errors in the conversion.

Thanks,

Scott