ARM-software/CMSIS_5 #199: osThreadJoin() with timeout

Masterbar, 2017-05-03:
HI,

Two years ago I requested an osThreadJoin() function and I'm happy that we have it now. But sometimes you don't want to wait forever for a thread termination. We need an osThreadJoin() function with a timeout.
This is useful when you shutdown your device and try to do it cleanly, requesting all the threads to exit (and wait their termination with osThreadJoin). But if any thread is blocked or looping forever you want to force its termination and then shutdown. That means we need a timeout on the join function.
You could have a look to the POSIX function pthread_timedjoin_np(). In POSIX you also have pthread_tryjoin_np which could be useful. Being closer to POSIX allows developer to easily port code to/from other systems.

Thank you,
Greg

JonatanAntoni, 2017-05-03:
Hi @Masterbar, 
Thank you for your proposal. I will discuss this enhancement and report back as soon as possible.

ilg-ul, 2017-05-03:
strictly speaking, `pthread_timedjoin_np()` is not part of POSIX.

POSIX does not really need timed join, since it relies on a reliable thread cancellation mechanism, so instead of a timed join, you kill the thread and then use a regular join(). the actual kill might be performed on the idle thread, so on very busy system it might not be immediate, but it should not hang. (if the scheduler does not reach the idle thread in 2-3 ticks, personally I would say that the system design has a problem, most probably a busy wait somewhere).

a reliable thread cancellation method might be useful for CMSIS RTOS too.

Masterbar, 2017-05-03:
@ilg-ul Thank you for the clarification about POSIX Liviu.
As we don't have this reliable thread cancellation method in CMSIS RTOS, don't yo think a timed join is useful? Or what would you propose?

What I'm trying to do is something like:
```
void shutdown(void) {
threadA->stop(); // This set a bExit boolean so that the thread can exit its main loop
osThreadJoin(threadA->tid);
threadB->stop(); // This set a bExit boolean so that the thread can exit its main loop
osThreadJoin(threadB->tid);
// ... other threads
fsumount();
// ...
}
```

I call this shutdown() function from a low priority thread. But I need to shutdown quickly so that I can umount the fs before my capacitor is empty...
The same function is also called for a reboot, in that case I need to be sure none of the child thread would block me forever.

ilg-ul, 2017-05-03:
> As we don't have this reliable thread cancellation method in CMSIS RTOS ... what would you propose?

I think that a reliable thread interruption/cancellation/killing would be more useful.

however, please note that thread cancellation may be slightly different from thread killing. thread killing is like sending signal 8 to a POSIX process; the process is killed, no questions asked.

thread cancellation is more like an orderly thread termination, you ask the thread to terminate, if the thread was blocked it is resumed with a special error code, the thread interprets this code, does a cleanup and then calls exit().

so, assuming you want to do a timed join, you first try to cancel, wait a timeout, then kill.

in CMSIS++, a `thread->kill()` is instant; a `this_thread.exit()` cannot be instant, because it cannot free the stack that is currently used, so it enqueues a job to the idle thread, which will do the `kill()`, also instant.

in other words, killing another thread is instant, killing itself requires a trip to the idle thread. if this is a problem, I guess the priority of the idle thread can be temporarily raised.

in CMSIS++ thread cancellation is currently done with `thread->interrupt()`, and checking with `thread->interrupted()`. thread interruption is not fatal, the thread can decide to continue, it just clears the interrupted flag. for terminal cancellation I have `thread->cancel()` (but the implementation details are to be specified; probably similar with `interrupt()`, except that clearing the interrupted flag will not be possible)

JonatanAntoni, 2019-03-12:
@RobertRostohar: Can we conclude on this one, i.e. either accept it as enhancement or decline it?