ARM-software/CMSIS_5 #200: Problem with Recursive mutex

juliobc, 2017-05-12:
Hi. 
I am using a recursive mutex in my app and I saw that it is running well but in a random time the funtion osMutexAcquire() returns != osOK when it is called from the owner thread, and then the aplication fails.

I only use osMutexRecursive attibute, not osMutexPrioInherit and not osMutexRobust.

What is the funtionality of osMutexRobust?? Some clue?

juliobc, 2017-05-12:
I saw that Lock Counter is 0xFF when the problem exists

juliobc, 2017-05-12:
Sorry, I did not know that the mutex had to be released every time it was acquired by the owner every time. Solved

JonatanAntoni, 2017-05-12:
Hi @juliobc. Thank you for your feedback.
I think our documentation has room for improvement here.
I am going to add some more details and short examples to the Mutex Flags. Thus I reopen this issue.
It would be nice if you could give it a review.

JonatanAntoni, 2017-05-12:
Please find additional documentation attached to the mutex macro definition.
Does this makes things clear?

juliobc, 2017-05-12:
Hi @JonatanAntoni, 
Great. 
Thank you

JonatanAntoni, 2017-05-12:
Please close the issue if you do not have additional feedback.

closed 2017-05-12 by juliobc