ARM-software/CMSIS_5 #201: rtx 5 supported iar version

YarivCol, 2017-05-18:
what is the minimum version of iar that support compiling rtx 5 ?

JonatanAntoni, 2017-05-19:
Hi @YarivCol,
with the current versions of IAR 7 and 8 it  should work, basically.

It might depend on the exact device (i.e. core class) you want to compile for.
The only hard requirement is that it provides recent CMSIS5 headers, so please refer to the IAR documentation. 

Do you have any special requirements on versions and targets you want to use?

Regards,
Jonatan

YarivCol, 2017-05-19:
i am trying to compile rtx 5 in iar v7.5 to stm32f407g, my problem seems to be the usage of __attribute__ around the code .
the svc defines in rtx_core_cm.h errors about syntax errors and missing ;
also __attribute__((section("TEMP"))); results in compile error.

JonatanAntoni, 2017-05-19:
We had various issues with RTX and IAR in the past, i.e. #152 and #149. Using IAR EWARM 7.80.4 it should work without any workaround. 
The usage of __attribute__((force_inline)) might be solved by using IAR _Pragma instead. 
Would you mind to try creating a fix that works for you and let us know, i.e. by issuing a pull request? That would be very helpful.

JonatanAntoni, 2017-05-29:
Hi @YarivCol,
Did you have the chance to either try using IAR 7.80.4 or modifying the pragmas?
Does one of these solutions work for you?
I would appreciate a final feedback so we might close this issue.

YarivCol, 2017-06-06:
hi, 
I compiled rtx with IAR 7.7 successfully!.
I didn't look at the release notes between  7.5 to 7.7 but i think they updated cmsis_iar.h along the way.

thanks for the help and quick response.

JonatanAntoni, 2017-07-17:
@YarivCol, I am closing this issue due to no further questions to be discussed. Please feel free to reopen if you still have any concerns.

closed 2017-07-17 by JonatanAntoni