ARM-software/CMSIS_5 #203: compiler abstraction need __PACKED_UNION

AmberAussie, 2017-06-08:
Hi,

I'm migrating code from keil compiler to llvm using cmsis_compiler.h and I face the issue with a "__packed union" inside a __packed struct.

It would be good if cmsis_compiler.h also define a __PACKED_UNION.

Best regards

JonatanAntoni, 2017-06-08:
Hi,

can you give us a reproducing example for what your are trying to do? Which exact compiler versions do you use?

Do you mind to provide the enhancement you have in mind as a pull request?

Kind regards,
Jonatan

AmberAussie, 2017-06-08:
Hi,

My code with ARM compiler 5 was:

```
typedef __packed struct {
    uint8_t type;

    __packed union {
      __packed struct {
        uint16_t len;
        uint8_t data[];
      } data;
      __packed struct {
        uint32_t cmd;
      } command;

    } payload;
} T_MESSAGE;
```

Now I try to migrate to ARM compiler 6.6 (which seem to be based on llvm). The idea is to be able to compile code using both compiler with the usage of cmsis_compiler.h but for the union there is no __PACKED_UNION to solve the issue.

```
#include <cmsis_compiler.h>
typedef __PACKED_STRUCT {
  uint8_t type;

  union __attribute((packed)) {
    __PACKED_STRUCT {
      uint16_t len;
      uint8_t data[];
    } data;
    __PACKED_STRUCT {
      uint32_t cmd;
    } command;
  } payload;
} T_MESSAGE;
```

I will do a pull request. For example in "cmsis_armclang.h" I will add:

```
#ifndef   __PACKED_UNION
  #define __PACKED_UNION       union __attribute__((packed, aligned(1)))
#endif
```

Thanks

JonatanAntoni, 2017-06-08:
That sounds reasonable to me. I'll take a look into and report back.
Thank you for your contribution.

JonatanAntoni, 2017-06-09:
Great. May I ask you to review the merged result and close this issue if solved, please?

closed 2017-06-09 by AmberAussie

AmberAussie, 2017-06-09:
It's fine. Thanks