ARM-software/CMSIS_5 #205: CMSIS-Driver: Dynamic Pin Layout

Microtronic, 2017-06-12:
We use the CMSIS-Driver for SPI/I2C communication inside our project.
Our application will run on 2 different hardwares. We would like to use the same binary
for both devices. We can detect the current hardware, but it is not possible to change the pin / ports of the CMSIS-Driver dynamically. Because the different hardwares havn't exactly the same pin layout.

The pin/ports are set with #defines and therefore of corse it isn't possible to change anything dynamically.

For our proposal it would be great if we could adjust the pin/ports after startup but before we call the initialize function _(Driver_SPI1->Initialize(...))._ This could be achieved if e.g. the resource object isn't defined static const.

```
// SPI1 Resources
SPI_RESOURCES SPI1_Resources = {
...
}
```

Then we have access to the pin layout with:
```

extern SPI_RESOURCES SPI1_Resources;
...
SPI1_Resources.io = /* new pin config */
```

Or if the CMIS-Driver has some function to access the resources object or to adjust the settings:

```
// with resource object access
*SPI_RESOURCES SPI1_GetResources(void){return &SPI1_Resources; }
// or with settings access
int SPI1_ChangePinLayout(...){return SPI1_ChangePinLayout(...,&SPI1_Resources); }
```

joabeck, 2017-06-12:
This is the reason why we cannot use CMSIS drivers at all in our application. We use the same binary in about 20 different but very similiar products. Some of them use different pins for the same peripheral. 
So if there is a way to choose the pins at runtime, the CMSIS drivers could be used more flexible.

One idea would be to give the resource object as a parameter to the init function, so that it is possible to have different static resource parameters for the different hardwares.

JonatanAntoni, 2017-06-12:
Hi @Microtronic,

which driver implementations do you refer to, i.e. which device(s) or family pack(s) are you using?

In the first instance CMSIS Driver only defines the interface. Your issue sounds like a detail of the concrete implementation you use.

Without knowing any further details I can imagine three solutions on the short run:
- Does the driver support skipping static pin configuration, i.e. does it accept NULL for the pins in the static structure? In this case you can freely take over responsibility for pin mux configuration manually.
- Otherwise you might change the pin muxing of your device AFTER calling Initialize. This solutions assumes the driver does not use the pins elsewhere during runtime. Unfortunately initialize temporarily configures a wrong pin mux which might be undesired. 
- As a last resort you can fork your customized drivers off the ones provided within the DFP in order to enhance them according to your requirements. You still benefit from the stable API other middleware components can relay on.

Please let me know if one of the solutions works for you. Or please provide me some more details.

Kind regards,
Jonatan

Microtronic, 2017-06-13:
Hi Jonatan
Wow, that was quickly answer. I didn't expected that 👍. 

I don't exactly know the version. There are so many different... (Main CMSIS version is 5.0.1)
And for example our SPI driver header is:
```
/* -----------------------------------------------------------------------------
 * Copyright (c) 2013-2015 ARM Ltd.
    ...
 * $Date:        06. Decemberber 2016
 * $Revision:    V2.9
 *
 * Driver:       Driver_SPI1, Driver_SPI2, Driver_SPI3,
 *               Driver_SPI4, Driver_SPI5, Driver_SPI6
 * Configured:   via RTE_Device.h configuration file
 * Project:      SPI Driver for ST STM32F4xx*/
```
Our application should run on a _STM32F437**I**IT_ and a _STM32F437**V**IT_.

Thank you very much for you workarround suggestions, but:
- The driver implementation doesn't support a NULL value. (I didn't know, but I had checked it right know)
- Of corse it would be possible to override the pin configuration after I call the _initialize_ function. We don't use the same pins for multiple use cases therefore these shouldn't be a problem. But it is not  really a proper solution, isn't it?
- I'm not really intersted to fork the driver impementation. In that case I can code my own driver functionality directly and without using the CMSIS driver. The CMSIS driver are great, because I don't have to care about a lot of things, but if I have to adjust the existing driver code, then it's easier to implement everything by my own.

Are there no future plans to add a CMSIS driver function to adjust the pin configuration?

Best regards
Microtronic

JonatanAntoni, 2017-06-13:
Hi @Microtronic,

thanks for the thumb up. :D
Good to know that our support provides a pleasant experience to you.

According to the file header you provided I suppose you are using the CMSIS Driver implementations shipped with STM32F4 Series Device Support Pack (Keil.STM32F4xx_DFP.2.11.0.pack). I will clarify with the development team maintaining this implementation what we can do.

Currently we do not have plans to enhance CMSIS Driver to adjust pin configurations. The main issue here is that pin configuration is very device specific. The CMSIS Driver APIs are completely device agnostic. We have to assure that the API can be implemented for a wide range of ARM-based microcontrollers. The major requirement for us to start standardizing the hardware driver APIs was to make middleware components easy portable and maintainable.

Due to the fact that the driver configuration structures are device specific (in your case they depend on the HAL provided by ST) we cannot simply make them accessible through the CMSIS Driver API. Thus I guess it is a matter of the concrete driver implementation which configuration options it offers. The easiest one which we think should cover the major part of user requirements is totally fine using a static compile time configuration (i.e. given in RTE_Device.h if you use MDK as the development environment). 

Please give me some time to think about possible solutions to support dynamic runtime configurations.

Kind regards,
Jonatan

JonatanAntoni, 2019-03-12:
@ReinhardKeil: Can we conclude on this one, i.e. either accept it as enhancement or decline it?

ReinhardKeil, 2019-03-12:
The issue is a one-off request.  While the idea behind is great, we cannot address every user aspect of CMSIS-Drivers. Today the CMSIS-Drivers are configured using CubeMX, and to my knowledge CubeMX does also not support dynamic pin configuration at run-time.

closed 2019-03-12 by ReinhardKeil