ARM-software/CMSIS_5 #206: CMSIS-Driver C++

jankap, 2017-06-16:
Hello,

the goal of our development team is to move to C++ in mid- to long-term time frame and at the same time we want to rely more on standards/APIs. A discussion is going on if we want to use the CMSIS-Driver API and either wrap it by C++ or implement the driver+API itself in C++ and provide some compatibility layer in C if ever needed. Both "solutions" do not seem to be reasonable, because the CMSIS-Driver API introduces an instance pointer to introduce some object like behaviour. This is exactly what C++ already has under the hood.

My question: How would you recommend to integrate these Drivers into a C++ enviroment? Or is there even a C++ Interface planned?

Thank you very much.

JonatanAntoni, 2017-06-19:
Hi @jankap,

Adding/integrating a C++ API to CMSIS is on our backlog. Unfortunately we do not have enough free capacities here to address this on the short run. I cannot even promise a time frame. But I would like to encourage you to keep working on this topic. We would be very appreciative if you provide us usage examples and implementation proposals taken from your real world applications.

The current CMSIS C-API is a simple struct with function pointers which could be seen like a virtual function table. A majority of CMSIS Driver implementations is simply a thin wrapper around vendor provided HAL libraries. The same could be applied using a C++-API. In these cases wrapping the C wrapper by an additional C++ wrapper would just append a second indirection.

In situations where the CMSIS Driver API is implemented natively we should think about a solution which does not need a wide duplication of code. Duplicated code would increase the maintenance effort needed. With CMSIS we want to support both, plain C and C/C++ applications. Thus we might come up with splitting the C implementations in a way that major parts can be reused by a C++ wrapper. But the driver code must still be usable in a plain C environment for users without any C++ possibilities.

Thank you for your contribution and please let us know when you are heading forward.

Kind regards,
Jonatan

jankap, 2020-11-05:
Dear @JonatanAntoni ,

our C++ migration takes much longer than expected due to a million of reasons. However I was also asked to evaluate Mbed's API & drivers in the near future.

Questions:
- Is there any development on the current C CMSIS-Driver API ongoing? Are they stable and up to date and widely used or will they (given the recent developments with Mbed) become deprecated soon?
- Any updates about a C++ variant?

Thanks

ilg-ul, 2020-11-05:
I did not check the latest version, but if you want to use the drivers in C++ be sure you have a way to pass the pointer to the object. For detail see my previous comment from 2017.

JonatanAntoni, 2020-11-05:
Hi @jankap,

we still have no plans to switch entirely to C++. The CMSIS-Drivers are considered mature and implemented for a variety of devices, see the Device Family Packs. The HAL/driver layer in mbed is using C-APIs as well. This is still common sense in order to leave the language choice to the application developers.

I think it should basically be not a big issue to use C-APIs (CMSIS-Driver) in an C++ environment. Please let me know about the most important ones of your million of reasons. I am happy to help.

Cheers,
Jonatan