ARM-software/CMSIS_5 #207: Address of SCS in CMSIS not defined in ARM Cortex-M7 Generic User Guide

mguntli, 2017-06-21:
# Summary
There is an issue regarding a documentation mismatch about the synonym *SCS (System Control Space)*.
But if even the simplest base addresses do not match between CMSIS and the ARM Cotrex-M7 Generic User Guide, what should be the reference then?

# Problem
The SCS_BASE address is used in CMSIS, but the synonym "SCS" does not exist in the ARM Cortex-M7 Generic User Guide (ARM DUI 0646B).
Of course I know that the reason that CMSIS the SCB does not match to the ARM User Guide is to not blow up the entire structure with 3320 reserved bytes, but it still is implemented in a very dirty way and the naming is a quite a mess..

# Core_CM7.h
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/core_cm7.h
* SCS_BASE is defined at 0xE000E000
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/core_cm7.h#L1754
* SCB_BASE is defined at 0xE000ED00
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/core_cm7.h#L1761
* SCnSCB is defined at 0xE000E000
https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/Core/Include/core_cm7.h#L1763

# ARM DUI 0646B:
http://infocenter.arm.com/help/topic/com.arm.doc.dui0646b/DUI0646B_cortex_m7_dgug.pdf
* A search for "System Control Space" returns 3 matches, but there is no address defined
* A search for "SCS" returns 2 matches, but there is no address defined
* Table 4-1 states that the System control block address range is 0xE000E008-0xE000E00F
* Table 4-12 lists 0xE000E008 as ACTLR (the first register of System control block)

mguntli, 2017-06-21:
I have created an example how I would have expected the SCB block to match with the ARM documentation:
https://github.com/mguntli/CMSIS_5/commit/cf37cecc8ec9f191308644ecffc328197de4e56e

JonatanAntoni, 2017-06-21:
Hi @mguntli,

Thank you for this feedback.

If you take a look into the header files for other Cortex-M type core registers you may notice that we use a similar structure everywhere. Due to the wide spread usage of CMSIS Core changes must be contributed very carefully. Moving a register from one struct to another would introduce a breaking change.

I admit that this difference between the user guides and the actual implementation might be confusing at the first glance. Do you have other concerns about the current implementation? Can we overcome these shortcomings by clear statements in the associated CMSIS Core(M) documentation? Is something missing you would like to have there?

I appreciate any feedback regarding beneficial improvements.

Kind regards,
Jonatan

mguntli, 2017-06-22:
I just thought: if ARM CMSIS **doesn't even match to the ARM user guide**, then what else?

Regarding backwards compatibility I do agree. If you don't remove ACTLR from SCnSCB_Type, then my proposed solution is 100% backwards compatible (except someone was relying on the **wrong base address of SCB_BASE**).
Personally I think I wouldn't take the risk to fix the wrong SCS_BASE for CMSIS, but it just gives me a bad impression that the whole CMSIS might not has been "coded with love" ;-)

# ARM DUI 0646B:
System control block = 0xE000E008 (Table 4-1 Core peripheral register regions)

# CMSIS:
SCB_BASE = 0xE000ED00

JonatanAntoni, 2017-06-22:
Thanks for sharing your sentiment.

I totally comprehend your statement. Despite all love we must be pragmatic. In this case we have had to decide to split the SCB or to let it span a huge memory range. So be sure that we are completely behind CMSIS even if we don't get every detail to its best in the limited time we can spent.

ReinhardKeil, 2017-06-27:
From the programmer's perspective we did not overlap various peripheral groups in the typedef's as this may become confusing. http://www.keil.com/pack/doc/CMSIS/Core/html/regMap_pg.html describes the actual register mapping from CMSIS to Technical Reference Manuals (TRM). As you can see there are several mis-matches which are caused by:
  - The various TRMs are not consistent across cores. CMSIS is consistent from the programmer's view.
  - Before CMSIS there were already established register names used by programmers (again different from TRMs). We have adopted that names to allow easier transition.
  - SCnSCB->ACTLR is according TRM part of SCB, but the struct would overlap with NVIC, SysTick, ect. if it would be defined that way.

I hope this helps you to under stand the deviations between TRMs and CMSIS.

mguntli, 2017-06-30:
Thank you Reinhard for your comment. It makes now more sense to me, that CMSIS tries to harmonize the "mess" of all the inconsistent ARM TRM for all ARM cores!

closed 2017-06-30 by mguntli