ARM-software/CMSIS_5 #208: core_cm7.h: Avoid duplicated code in cache management

mguntli, 2017-06-21:
The following functions could be refactored to avoid duplicated code. I am sure there is much more, but this is just an example:

* SCB_EnableICache
https://github.com/ARM-software/CMSIS_5/blob/master/CMSIS/Core/Include/core_cm7.h#L2193
Line 2196-2200 is basically duplicated code of SCB_InvalidateICache..
* SCB_InvalidateICache
https://github.com/ARM-software/CMSIS_5/blob/master/CMSIS/Core/Include/core_cm7.h#L2229

This of course means that the SCB_InvalidateICache has to be placed on top of SCB_EnableICache.
Was there a specific reason to not reuse existing functions?

JonatanAntoni, 2017-06-21:
Thank you for your feedback. I cannot proclaim a specific reason for the code duplication.

In this case the code fragment you point to consists of five lines. Four of them are explicit memory barriers and one is a simple register write. Which notable benefit (e.g. in terms of readability or maintainability) do you expect by introducing a further indirection (i.e. function call) in this case?

I admit one might use a function instead of the duplicated register writes. But using a single function would make it harder to decide which barriers are really needed because of the missing context. Do you have concerns to stick with the duplicated code in this very limited case?

mguntli, 2017-06-22:
Jonathan you are absolutely right. I don't want to start a debate about coding style, I was actually curious if there was an actual reason:
- it can't be performance: the function declaration is __STATIC_INLINE
- in terms of readability, I prefer a small function over 5 lines of code with comments, but that's maybe just me

In terms of "reading-flow" this is much more readable:

```
__STATIC_INLINE void SCB_EnableICache (void)
{
  #if defined (__ICACHE_PRESENT) && (__ICACHE_PRESENT == 1U)
      SCB_InvalidateICache();
      SCB->CCR |=  (uint32_t)SCB_CCR_IC_Msk;  /* enable I-Cache */
      __DSB();
      __ISB();
  #endif
}

```

```
__STATIC_INLINE void SCB_EnableICache (void)
{
  #if defined (__ICACHE_PRESENT) && (__ICACHE_PRESENT == 1U)
    __DSB();
    __ISB();
    SCB->ICIALLU = 0UL;                     /* invalidate I-Cache */
    __DSB();
    __ISB();
    SCB->CCR |=  (uint32_t)SCB_CCR_IC_Msk;  /* enable I-Cache */
    __DSB();
    __ISB();
  #endif
}
```

closed 2017-06-22 by mguntli